arduino/arduino-cli #1006: Library path changed in new windows 10 update

Conplug, 2020-10-04:
## Bug Report
Building failed by libraries losing.

### Current behavior
I run 
`arduino-cli lib  list`
`No libraries installed.`

Checked the windows's document path, it was changed to
`C:\Users\[my name]\OneDrive\Documents\Arduino\libraries`

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.13.0 Commit: 693a045
- OS and platform: Windows 10 Pro

### Additional context

ubidefeo, 2020-10-05:
hi @Conplug 
I think you chose to have all your documents stored on OneDrive and this changed your path.

@arduino/team_tooling I hope this does not turn into a nightmare 🤦‍♂️

alranel, 2020-10-05:
I'm not sure this should be handled anyhow in arduino-cli, as that path is entirely provided by the system :)
The easiest solution would be to move your Documents\Arduino folder to the new path, according to your new system configuration.

ubidefeo, 2020-10-05:
@alranel 
this requires editing the `arduino-cli.yaml` file which is in the `AppData` folder, and it's the default path for our data folder.
The problem is that the default `arduino-cli.yaml` created in the `init` process will point to the standard user path, and will not contain `OneDrive` or account for it.

@Conplug could you edit the `arduino-cli.yaml` file to point to your new path anywhere necessary?
these are the entries you're interested in for such operation (mine are for Mac OS but you'll be able to change yours easily)
```
directories:
  data: /Users/ubi/Library/Arduino15
  downloads: /Users/ubi/Library/Arduino15/staging
  user: /data/Dropbox/AppsDocuments/Arduino
```

alranel, 2020-10-05:
Ah, didn't know we store the path during init. In this case I see two possible changes to consider:

* do not store paths at all during init, and recompute them at every run if the keys are not found in arduino-cli.yaml (i.e. if the user didn't set custom ones manually)
* in case the stored paths do not exist, try to recompute them on the fly as a fallback

I think the first option is more correct: if we rely on a system-provided path, we should not consider it permanent but we should continue relying on the system providing the path. I don't know if there are implications in arduino-cli, but suggesting this as it's a common pattern :)

mastrolinux, 2020-10-07:
Also, consider the Documents dir is protected from external modification in most recent Win10 versions.

kittaakos, 2020-10-07:
> Documents dir is protected

Related: https://github.com/arduino/arduino-cli/issues/566

matthijskooijman, 2020-10-08:
One other option to solve this would be to allow windows variables in the config (e.g. something like `%USERDIR%` (or whatever that is called exactly,  I think such standardized variables exist)?

Yet another (devil's advocate) way to look at this, is that by moving your Documents folder into OneDrive,the user moved Arduino-cli's files, so it's up to them to tell arduino-cli where the files went. If I rename the `Arduino` folder to something else, arduino-cli cannot autodetect this, so why would this change need to be handled?

ubidefeo, 2020-10-08:
@matthijskooijman 
Before beginning to work on this project I thought Apple and their walled gardens (which I'm comfortable with) were hindering developers in many ways, but the amount of issues I see popping up because of Windows' odd behaviours if baffling to say the least 😂

umbynos, 2022-11-25:
> Checked the windows's document path, it was changed to

@Conplug what do you mean by "it was changed to"? You changed it?