arduino/arduino-cli #1007: feature request: add board name string to compile defines in platform.txt file

bperrybap, 2020-10-05:
Currently there is a define for the board type -DARDUINO_{build.board} that is passed to the compiler.
But the actual name of the board as a string is not available.
There are some use cases in diagnostic tools where having the actual name of the board as a string can be useful as it can be used to display to the user.
The esp8266 core uses this define -DARDUINO_BOARD="{build.board}" in their platform.txt file.
It would be nice if the Arduino.cc provided cores supported a define that allowed the code to obtain a string version of the board name.
Since the name ARDUINO_BOARD is already used by the esp8266 core, perhaps the Arduino.cc cores could adopt the same thing.

per1234, 2020-10-05:
Related to #985

bperrybap, 2021-03-30:
Why was this closed?
Are issues close automatically by a bot without human intervention after some amount of time?

per1234, 2021-03-30:
Hi @bperrybap. I apologize for the confusion. I seem to have broken the "stale-bot" workflow somehow. This was never intended to be closed.