arduino/arduino-cli #1011: [Proposal] Declare library depencies inside sketches

philippejadin, 2020-10-05:
This proposal is short and addresses a shortcoming for every user : which library do I need to install to make this sketch run?

- allow to add some markup to sketches to declare the required library and version needed 
- parse this new info on compile, and ask user if he wants to install the needed libraries
- profit!

**Markup proposal, example with the rfid lib :** 
```
// require_arduino_lib "MFRC522"
#include <MFRC522.h>
```

Note that we simply use a comment so nothing changes from the compiler point of view.

**Markup proposal specifying the lib version to use :** 
```
// require_arduino_lib "MFRC522" "1.4.2"
#include <MFRC522.h>
```

The rules are as follow : 
- the word `require_arduino_lib` 
- followed with the quoted  library name
- followed by the quoted version or nothing to install the latest
- followed by a new line character
- one library per line
- inside a comment is allowed (and currently mandatory since there is no compiler support)

ianfixes, 2020-11-30:
The easiest workaround I know of is to convert your sketch into one of the "examples" for a [library](https://arduino.github.io/arduino-cli/latest/library-specification/).  Then you can put the dependencies in `library.properties`

```
name=WebServer
version=1.0.0
author=Cristian Maglie <c.maglie@example.com>, Pippo Pluto <pippo@example.com>
maintainer=Cristian Maglie <c.maglie@example.com>
sentence=A library that makes coding a Webserver a breeze.
paragraph=Supports HTTP1.1 and you can do GET and POST.
category=Communication
url=http://example.com/
architectures=avr
includes=WebServer.h
depends=ArduinoHttpClient
precompiled=true
```

> note the line `depends=ArduinoHttpClient`

The library itself can be basically empty, e.g. [do-something.cpp](https://github.com/Arduino-CI/arduino_ci/blob/master/SampleProjects/DoSomething/do-something.cpp) from the [dummy `DoSomething` library](https://github.com/Arduino-CI/arduino_ci/blob/master/SampleProjects/DoSomething/)

tarampampam, 2021-06-18:
@philippejadin Maybe https://github.com/arduino/arduino-cli/issues/1321 can be helpful for you?

umbynos, 2022-10-12:
Actually this has been inplemented in the cli https://arduino.github.io/arduino-cli/latest/sketch-project-file/.

closed 2022-10-12 by umbynos