arduino/arduino-cli #1012: Uploader prefs miss  {build.core.path}, {build.system.path} and {build.variant.path}

jandrassy, 2020-10-06:
The keys {build.core.path}, {build.system.path} and {build.variant.path} are not evaluated for platform.txt tools recipes (tools.xy.upload.pattern).

Then the tools recipes must use {runtime.platform.path}. But then there are problems to use the package as referred, because {runtime.platform.path} is the referring package path (where the boards.txt is).

For example in samd package the upload.pattern contains `{runtime.platform.path}/variants/{build.variant}/{build.openocdscript}` instead of `{build.variant.path}/{build.openocdscript}`. A board in referring package can use in board.txt an existing variant of the referred package in `xy.build.variant` for compilation, but the uploader will not find the script for the variant.

I try to make the esp8266 package referrable. Now it uses `{runtime.platform.path}/tools` all over platform.txt and `{runtime.platform.path}` will evaluate to referring package, not to referred package. They could use `{build.core.path}/../../tools` or rename the folder to `system` and use `{build.system.path}` variable. It works for building, but it doesn't evaluate in uploader.pattern.

I debugged the Arduino IDE and found out that compiler recipes use prefs evaluated by arduino-builder, while Uploaders use the preferences map of the IDE. So it is not a simple fix. I think, the best way for fixing this would be to pass the prefs evaluated by `Compiler` over `SketchController` from build() to upload(). Same in class `processing.app.Base`.

jandrassy, 2020-10-06:
@per1234 Per, now I know that the upload in IDE is still the old built-in one. I didn't try upload with CLI.
should I close this?

per1234, 2020-10-06:
I did a test with the latest nightly of Arduino CLI before moving this and those properties were not expanded in the upload recipe. So I don't think your feature request is resolved. If I'm wrong about that, and it is now supported, then it should be closed as resolved.

It's the Arduino CLI codebase where this feature request would be implemented, which will then eventually become available to all the tools that use Arduino CLI, including the IDE. So it now makes more sense for this issue to be in the `arduino/arduino-cli` repo than the IDE repo.

jandrassy, 2022-01-11:
the issue is not resolved in IDE 2.0

jandrassy, 2022-01-14:
after esp8266 and esp32, [the pico core ](https://github.com/earlephilhower/arduino-pico) is next platform which stockpiles all boards because it can't be referred because of this problem. and custom boards definitions in sketchbook's hardware folder have to copy and maintain files used in upload recipes.

[the specification](https://arduino.github.io/arduino-cli/0.20/platform-specification/#tool-references) as written now makes this a feature, not a bug, but this problem practically stops use of referencing other core.
> Note that, unlike core references, referencing a tool recipe does not result in any other resources being inherited from the referenced platform.

Is it really not possible to make it work in CLI and IDE 2?

mcbp223, 2023-01-16:
While waiting for an implementation, can we symlink to the tools directory in the referenced package and expect the same behavior once a solution is adopted?

JAndrassy, 2023-01-20:
> While waiting for an implementation, can we symlink to the tools directory in the referenced package and expect the same behavior once a solution is adopted?

if you meant the tools folder of the esp8266 arduino platform, then yes the simplink is a good workaround.
you will still have to change to `runtime.tools.eboot` to `{build.core.path}/../../bootloaders/eboot/eboot.elf`

mcbp223, 2023-01-20:
> > While waiting for an implementation, can we symlink to the tools directory in the referenced package and expect the same behavior once a solution is adopted?
> 
> if you meant the tools folder of the esp8266 arduino platform, then yes the simlink is a good workaround. you will still have to change to `runtime.tools.eboot` to `{build.core.path}/../../bootloaders/eboot/eboot.elf`

Thanks for the hint on `{build.core.path}`, I haven't tried referencing the esp8266 platform yet but it's in the pipeline. I'm currently using the method to reference esp32, with a `platform.txt` file containing only the name of the collection, `boards.txt` with `esp32:` prefix additions to each board for `{build.core}`, `{bootloader.tool}`..., `{upload.tool}`..., and variant folder copied to the referencing package, to allow for customisations and library specific include files. There was too much to copy for the tools, a symlink worked for me better.
I tried referencing Earle's RP2040 core but no luck yet with tools location, I'll try what you suggested using a relative path. I've heard other people had success with stm32, and of course avr, planning on trying them. Looking to do more referencing and less copying, and eventually not having to make symlinks that need to be changed with each platform upgrade.

It's important to get this method to work, and be somewhat standardized and relatively easy to use, since it answers to years of requests from both users and library maintainers to allow libraries depend on custom user boards, does best to model reality (a combined `board + shields` has different specs than the ones shipped with the platform for just the board), and at the same time allows for a unidirectional dependency graph:
`dev board` -> `dev board + shields` -> `libraries` -> `sketch`

JAndrassy, 2023-01-21:
@mcbp223 I don't want to pollute the issue here with a discussion so just two links for you
https://github.com/esp8266/Arduino/issues/5769
https://github.com/JAndrassy/my_boards
If you want to discuss it more I am on arduino.forum.cc or open a discussion in my my_boards repository

PR #2082 (2023-03-02): Refactored build-properties creation (and removed some legacy code)
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [X] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [X] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug [... cut]
closed 2023-03-02 by cmaglie