arduino/arduino-cli #1035: Incorrect "Sketch too big" error on insufficient Flash space when uploading using programmer

deltaphi, 2020-10-18:
Given a sketch of compiled size X, where X by itself will fit into the flash of a given chip, but X plus the Arduino Bootloader will not. When using the regular "Upload" Button, the Arduino IDE correctly prints a warning about insufficient flash memory and aborts the upload process.

However, when using "Upload using Programmer" (precisely to use the extra space by getting rid of the bootloader), the Arduino IDE again prints this warning (this time incorrectly!) and aborts the upload process.

To resolve this issue, the IDE should take the upload method (Bootloader/Programmer) into account when computing the amount of flash available on the chip.

I verified this using Arduino 1.0.4 and Arduino 1.0.5, both times compiling for an ATmega8 ("Ardunio NG or older w/ATmega8").

Until this bug is resolved, I would also appreciate a hint on how to work around this issue (i.e., how to burn an "oversize" sketch with a programmer) in the meantime.

per1234, 2020-10-18:
avrdude at least supports programming fuses and flash in a single command:
```
tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} -Uhfuse:w:{program.high_fuses}:m  "-Uflash:w:{build.path}/{build.project_name}.hex:i"
```
so the only thing that would be needed is an property equivalent to [`upload.maximum_size`](https://arduino.github.io/arduino-cli/dev/platform-specification/#__code_11:~:text=%7Bupload.maximum_size%7D%3A%20available%20program%20storage%20space) for use with the `program` action.

The current workaround for this in the 3rd party platform world is to add a "no bootloader" custom board option ([example](https://github.com/MCUdude/MiniCore/blob/v2.0.7/avr/boards.txt#L74-L78))
but that's not quite so user friendly as the `program` action handling everything automatically.

deltaphi, 2020-10-18:
> > In my usecase, there actually is no boot section.
> 
> If you have a custom board that is _never_ intended to be used with a bootloader, the `boards.txt` for that board can both set the fuses (at least on AVR) to disable the bootloader and the max flash size can be set to the full flash size, and this problem will not exist (but uploading using the bootloader will not be available, unless there is some ROM bootloader that takes no space).

Thinking back to the question that raised this bugreport: I had a custom circuit soldered onto a perfboard, including the typical Arduino Uno parts (Xtal, Reset button, ISP Header, USB-Serial Adapter Header). I socketed store-bought Atmega328s into those boards. For the project I actually preferred not having the bootloader, since the 1 second delay at power on would have been annoying. Needing the flash space for the project was a problem on top, which eventually prompted this bugreport. The entire project was originally inspired by the "Arduino on Breadboard" and "DIY Arduino on Perfboard" post surrounding the Arduino community at the time.

As I had store-bought AVRs, I must have programmed the fuses somehow, as the factory defaults don't work for an Arduino Uno board. I remember running into the issue of "Upload using programmer" not setting the fuses and incorrectly writing up a whole bunch of the AVRs as defective. Although whether I set the fuses using avrdude directly, using a custom boards.txt or messing with an existing boards.txt, I cannot say anymore.

In any case, I was still working on on the code when I had set up circuit on the perfboard (removing bugs, adapting pinlayouts to circuit board revisions, adding the occasional functionality and the like). So this was not a matter of "mess with avrdude once" but of repeatedly recompiling my code and uploading through the programmer.