arduino/arduino-cli #1046: Feature request: arduino-cli compile --show-properties option to show fully expanded properties

obra, 2020-10-23:
When using arduino-cli as part of a larger build system, I often need to fetch build properties for a sketch at build or flash time.

`arduino-cli compile --show-properties` will give me a dump of all the properties I might possibly want, but it dumps them as non-expanded templates.

For example, the default avr `compiler.path` gets dumped as:

`compiler.path={runtime.tools.avr-gcc.path}/bin/`

I would find it hugely valuable to have an option to --show-properties that shows the properties after template substitution.

obra, 2021-03-30:
Can I ask why this got closed? Was the functionality implemented? Or are feature requests now being tracked somewhere else?￼￼

per1234, 2021-03-30:
Hi @obra. I apologize for the confusion. I seem to have broken the "stale-bot" workflow somehow. This was never intended to be closed.

obra, 2021-03-30:
No worries. I’ve got a few other similar tickets. I’ll reply to them later today

per1234, 2021-03-30:
I appreciate your understanding. I have now reopened the many issues and pull requests that were closed erroneously. If there are any I missed please do comment and I'll correct the omission immediately.

obra, 2021-03-30:
I think you got all of mine :)

Thanks for the quick response

closed 2023-02-17 by Bikappa