arduino/arduino-cli #1047: Feature request: `compile --show-properties` without a sketch

obra, 2020-10-24:
`arduino-cli compile --show-properties`  currently won't run without an `.ino` file present. However, that appears to be a side effect of the fact that it's an option to `compile`

Using arduino-cli in my tooling, I often want to query board properties before locating a specific sketch. (This may be crazy.)

I -sort of- feel like `--show-properties` should be a sub-command of `arduino-cli board`, rather than compile. (And, if we're going there, I'd love it if there were a command like:

```
./bin/arduino-cli board get-property foo.bar --fqbn arduino:avr:leonardo
```

which would return the fully evaluated template for foo.bar.

ubidefeo, 2020-10-24:
@obra 
this looks interesting.
What do you think about it, @per1234 ?

per1234, 2020-10-24:
I actually think it would be cool if it were a global flag, since the properties are context-dependent. If you are compiling a sketch, `build.project_name` and `build.source.path` will be defined. If you are uploading, `serial.port` and `serial.port.file`, if doing `upload -P` or `burn-bootloader` then the properties of the programmer are defined.

obra, 2020-11-15:
Just to further explain how I'm using this in my build tooling, these are the current shell functions I use to get an individual property.

```
_arduino_prop() {
        pref=$1
        # Strip the preference name. And then strip leading and trailing quotations
        _arduino_props | grep --max-count=1 "${pref}=" | sed -e s/^.*"${pref}"=// -e 's/^"//' -e 's/"$//'
}

_arduino_props() {
        if [ "x${_ARDUINO_PREFS}x" == "xx" ]; then
                _ARDUINO_PREFS=$(_run_arduino_cli --fqbn "${FQBN}" compile --show-properties "${SKETCH_FILE_PATH}")
        fi
        echo "$_ARDUINO_PREFS"
}

Here are some of my current uses:

 
``` 
# We have a target that builds a .hex including the sketch and the bootloader

"${BOOTLOADER_PATH:=$(_arduino_prop 'runtime.platform.path')/bootloaders/$(_arduino_prop 'bootloader.file')}"```

```
# A custom board property used in the flashing process

flashing_instructions=$(_arduino_prop 'build.flashing_instructions')```
```

umbynos, 2023-02-02:
Related to #1046

closed 2023-04-19 by cmaglie