arduino/arduino-cli #1051: [ Breaking Change in upcoming Nightly ] Sketch level build folder is not created as default

ubidefeo, 2020-11-02:
Dear brave testers and bleeding edge users, 
as time moves forward we introduce another breaking change in our next Nightly Build, so be wary of the changes and act accordingly.

If your workflow relies on picking files from the enclosed `build` folder you'll need to manually specify the output as per this PR https://github.com/arduino/arduino-cli/pull/1042#event-3947571860

Before you could rely on a `--dry-run` flag which was kind of semantically incorrect, since it produced artifacts.
Now you need to specify the `-e | --export-binaries` flag if you want the old behaviour to be observed, and this has also been turned into both a configuration parameter for `arduino-cli.yaml,` as well as an Environment Variable.

These are, respectively
`sketch.always_export_binaries` for the `sketch` section in the `arduino-cli.yaml` file and 
`ARDUINO_SKETCH_ALWAYS_EXPORT_BINARIES` for the ENV variable

These need to be set to `true` in order to take effect or `false` for the default behaviour.
If neither the `arduino-cli.yaml` parameter nor the Environment Variable are present, the default behaviour applies.

If your workflow already uses `--output-dir` you do not need to specify any of the above, because explicitly setting an output directory will carry such behaviour with and only copy files where requested.

This new behaviour reduce the need to manually add items to `.gitignore` and also reduces littering since a build folder is created in temp paths and those get flushed when the system needs to.

**example for `arduino-cli.yaml` entry**
```
sketch:
  always_export_binaries: true
```

ubidefeo, 2020-11-02:
@silvanocerza @per1234 @rsora please comment if you want anything to be changed

ianfixes, 2020-11-16:
What is the easiest / most straightforward way to check that the sketch can be compiled, and nothing more?  In other words, I'd prefer that the artifacts stay out of sight in some temp directory and never pollute the project workspace. 

How should I plan to achieve this?  Best-case scenario for me is purely via command-line args (no config file writing).

per1234, 2020-11-16:
>I'd prefer that the artifacts stay out of sight in some temp directory and never pollute the project workspace.

Starting from the next release of Arduino CLI, this will be the default behavior, so no special command will be needed.

With the current release version 0.13.0, you could use `arduino-cli compile --dry-run`, but that command will break on the next release due to `--dry-run` having been removed in https://github.com/arduino/arduino-cli/pull/1042. So I think the best approach is to use `arduino-cli compile --output-dir` with the output directory set to some temporary directory where the build output won't cause you any bother. Even though the `--output-dir` flag will no longer be necessary in Arduino CLI 0.14.0 and onward, it also won't cause any problems because the `--output-dir` flag has not been removed.

closed 2021-02-03 by per1234