arduino/arduino-cli #1058: Support for recursive search on compile --library flag

cSDes1gn, 2020-11-05:
Hey everyone,

Im working on an environment for building bigger arduino projects with a set of local dependancies located in the root `include` directory as shown below:
```
workspace/
│
├── include/
│   ├── Segment/
│   │   ├─── src/
│   │   │   ├─── Segment.h
│   │   │   ├─── Segment.cpp
│   ├─── package.h
│   └─── LICENSE
|   |
│   ├── Matrix/
│   │   ├─── src/
│   │   │   ├─── Matrix.h
│   │   │   ├─── Matrix.cpp
│   ├─── package.h
│   └─── LICENSE
├── src/
│   ├── src.ino
│
├── Makefile
```
The `Makefile` uses `arduino-cli compile` to compile the `src.ino` sketch and using the `--library` flag to point to `${PWD}/include/Segment` and `${PWD}/include/Matrix` seperately. I am trying to include all of the local libraries inside my `include` directory but it looks like the cli does not support a recursive search option for the `--library` flag. This is a problem for automation purposes since I don't want to have to manually add new paths to the command as the libraries grow.

Any plans on support for this type of feature? Any insight would be appreciated.

ianfixes, 2020-11-30:
Why wouldn't you just install (or even symlink) the libraries into the Arduino libraries folder (or even a custom one, since you can specify that from the CLI)?  The tool seems to follow the dependency chains in `library.properties`.

cSDes1gn, 2020-11-30:
> Why wouldn't you just install (or even symlink) the libraries into the Arduino libraries folder (or even a custom one, since you can specify that from the CLI)? The tool seems to follow the dependency chains in `library.properties`.

This is exactly how I have resolved the issue. Used `--libraries ${PWD}/include` as a compile flag and added the library.properties file to each of the include packages. Seems to work well for me thanks!