arduino/arduino-cli #1059: Bug Report: Can't install samd cores

arock093, 2020-11-05:
I thought about posting this in the forum. It seems like a underlying technical issue so hopefully this will be alright.

I'm using a raspberry pi zero w. Based off of reports by others in the forum, it should be able to handle arduino-cli.
I tested the command "arduino-cli core install arduino:avr" and it worked fine.
I'm trying to run the commands, arduino-cli core install arduino:samd and arduino-cli core install adaruit:samd.

This is the output. Both commands get hung up on "Installing arduino:arm-none-eabi-gcc@7-2017q4..." and then are Killed.

Downloading packages...
arduino:arm-none-eabi-gcc@7-2017q4 downloaded                                   
arduino:bossac@1.7.0-arduino3 downloaded                                        
arduino:openocd@0.10.0-arduino7 downloaded                                      
arduino:CMSIS@4.5.0 downloaded                                                  
arduino:CMSIS-Atmel@1.2.0 downloaded                                            
arduino:arduinoOTA@1.2.1 downloaded                                             
arduino:samd@1.8.9 downloaded                                                   
Installing arduino:arm-none-eabi-gcc@7-2017q4...
Killed

The process is getting killed for some reason. 

dmesg output indicates a memory issue:
Out of memory: Killed process 514 (arduino-cli) total-vm:938304kB, anon-rss:398624kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:598kB oom_score_adj:0

per1234, 2020-11-05:
Hi @arock093. Thanks for your bug report.

Please post the output from this command:
```
arduino-cli version
```

arock093, 2020-11-05:
> Hi @arock093. Thanks for your bug report.
> 
> Please post the output from this command:
> 
> ```
> arduino-cli version
> ```

arduino-cli Version: 0.13.0 Commit: 693a045

per1234, 2020-11-05:
OK, I think this bug might have been fixed since the time of the 0.13.0 release version you're using (https://github.com/arduino/arduino-cli/pull/981).

Please try again with the nightly build of Arduino CLI and then let us know if that fixes the issue:
https://arduino.github.io/arduino-cli/latest/installation/#nightly-builds

arock093, 2020-11-05:
> OK, I think this bug might have been fixed since the time of the 0.13.0 release version you're using (#981).
> 
> Please try again with the nightly build of Arduino CLI and then let us know if that fixes the issue:
> https://arduino.github.io/arduino-cli/latest/installation/#nightly-builds

I used the nightly build and it allowed me to install the cores. Now I'm having an issue compiling.

arduino-cli compile --fqbn adafruit:samd:adafruit_trinket_m0 MyFirstSketch
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.3/cores/arduino/WInterrupts.c: In function 'detachInterrupt':
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.3/cores/arduino/WInterrupts.c:378:1: internal compiler error: Illegal instruction
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.3/cores/arduino/cortex_handlers.c: In function 'Reset_Handler':
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.3/cores/arduino/cortex_handlers.c:543:1: internal compiler error: Illegal instruction
 }

ubidefeo, 2020-11-09:
@arock093 
unfortunately there will always be some issues with `armv6` because some tools (or versions of them) do not run happily on it.
If you move to `armv7` (Pi3) you should have no issues.
The Zero is also quite slow and runs on 512MB of RAM.
The tests I ran on it a while back took an excruciatingly long time to compile for SAMD and when it came to mbedos I could sit there forever :(

Right now my only hope when it comes to that format is that Raspberry Pi Foundation releases a Zero form-factor board with more RAM and a new CPU :)

arock093, 2020-11-24:
Thanks for the help and feedback everyone. I apologize for opening an issue when it was less of a software issue and more hardware compatibility problem.

I was able to get an Arduino Micro to work with the Raspberry Pi Zero. 

As far as this issue goes, should I close it and try to get the adafruit:SAMD repo to mention somewhere about the Pi zero having compatibility issues?

ubidefeo, 2020-11-24:
@arock093 
when it comes to 3rd party cores we don't have power, but we are working towards making sure that dependencies are properly installed when they are properly defined in the core package.
I haven't gone through Adafruit's ones, but I know they do their research and are very responsive when it comes to issues reports :)

umbynos, 2022-11-25:
Closing because the original Out of Memory bug was resolved. Thanks for the interest in the project :heart:

closed 2022-11-25 by umbynos