arduino/arduino-cli #1068: Feature request: board list --fqbn?

obra, 2020-11-12:
As I've talked about in a previous issue, I'm in the process of replacing a couple thousand lines of perl, bash, and powershell with a thin set of wrappers around arduino-cli.  For the most part, I'm really happy with how much code I get to delete. 

I've just hit a place where I'm not sure I see a clean way to do what I want.

Without depending on a json parser, I don't currently see a reasonable way to answer the question "what is the full path to the serial port for the connected board (or boards) with fqbn $FQBN?"

One way to get what I need might be if `arduino-cli board list` honored the --fqbn flag and used the result as a filter for the list of ports found.

I have an additional wrinkle that I'll bring up in a new related issue.

ubidefeo, 2020-11-15:
hey @obra 
I guess you want to flash multiple boards of the same kind, do you?

Seems like an interesting case for hardware/device makers, let me squeeze it in our backlog and discuss it with the team.
We do all these things using `jq`, but I guess your 
>Without depending on a json parser

rules that out 😀

obra, 2020-11-15:
@ubidefeo - Not....quite?

The end goal is to have a Makefile target of "make upload" for a sketch that "just works" for users, assuming we know the FQBN. If a json parser ends up as a hard requirement, I understand, but I'm hoping it won't, as one of the nice things about arduino-cli is how self-contained it is. 

Like the leonardo and micro, our boards have different USB pids for the bootloader and sketch, so we can easily determine if the board needs to be reset into the bootloader and can prompt the user to hold in a key on the board to stay in the bootloader and accept an upload.

The way we did that in the arduino-builder days is:

* use the FQBN to ask arduino-builder for the vid/pid pairs for the board
* use a horrible platform-specific script to go digging for all the serial ports for that vid/pid pair.
* if the serial port has the sketch pid, display a message and wait for the user to confirm
* reset the board into bootloader mode 
* find the serial port again
* go ahead and upload

Parsing the non-json output of board list -sort of- works, but feels very brittle and like something you probably don't want me doing.  If two different boards have the same vid/pid, both show up in the non-json board list, but the path to the serial port only shows up for the first one listed.   

This is a slightly simplified version of what I've currently got. Ideally, I'd also be able to see the PID of the current board, so I can skip prompting the user if they've already got the board in bootloader mode. (Although `arduino-cli upload` does deal gracefully either way)

```
port=$(arduino-cli board list --format=text | grep "$FQBN" |cut -d' ' -f 1)
flashing_instructions=$(_arduino_prop 'build.flashing_instructions')
: "${flashing_instructions:="If your keyboard needs you to do something to put it in flashing mode, do that now."}"

printf '%b\n\n' "${flashing_instructions}"
echo "When you're ready to proceed, press 'Enter'."
read
arduino-cli upload --fqbn "${FQBN}" --port "${port}" 

```

_arduino_prop looks like this:

```
_arduino_prop() {
        pref=$1
        # Strip the preference name. And then strip leading and trailing quotations
        _arduino_props | grep --max-count=1 "${pref}=" | sed -e s/^.*"${pref}"=// -e 's/^"//' -e 's/"$//'
}

_arduino_props() {
        if [ "x${_ARDUINO_PREFS}x" == "xx" ]; then
                _ARDUINO_PREFS=$(_run_arduino_cli --fqbn "${FQBN}" compile --show-properties "${SKETCH_FILE_PATH}")
        fi
        echo "$_ARDUINO_PREFS"
}

```

umbynos, 2022-11-25:
Hi @obra, thanks for your interest in your project.
We accept your use-case, we added this to our milestone 1.0

closed 2023-01-30 by Bikappa