arduino/arduino-cli #1070: [feature request] preprocessor define in config (or on command line directly)

i-make-robots, 2020-11-13:
Thank you for your great work on arduino-cli.  I really appreciate all your time & energy.

I would like to feed arduino-cli a matrix of AVR types and machine styles to test them all, automatically, in a Github Action.
I have https://github.com/MarginallyClever/Makelangelo-firmware/edit/master/.github/workflows/main.yml
Many AVRs, many kinematic models run by one firmware.
kinematic models are decided with a preprocessor directive. for instance,
```#define MACHINE_STYLE = POLARGRAPH````

Please!  In a perfect world I would like to call
````arduino-cli compile --fqbn arduino:avr:mega ./Makelangelo-firmware.ino --define MACHINE_STYLE = POLARGRAPH````

If this is already possible I do apologize, I was unable to find the documentation.

obra, 2020-11-13:
#846 and #1044 are at least related. 

extra_flags and --build-property may give you the rope you need.

per1234, 2020-11-13:
Closing as duplicate of https://github.com/arduino/arduino-cli/issues/159

closed 2020-11-13 by per1234

i-make-robots, 2020-11-13:
Oh awesome.  I didn't find the documentation for this (fairly new) feature.  Sorry to bother.  Stay Excellent!