arduino/arduino-cli #1073: Allow specifying FQBN *and* port in `board attach`

matthijskooijman, 2020-11-14:
## Bug Report

### Current behavior

The `board attach` command allows specifying a port *or* FQBN. If you specify a port, the FQBN is autodetected from the board (if possible) and the resulting sketch.json has both the port and fqbn set. If you specify just the FQBN, only that is set in sketch.json.

For boards that *can* be autodetected, it is sufficient to specify the port. But for boards that *cannot* be autodetected (i.e. that use a common USB-to-serial chip without their own VIDPID), specifying the serial port is now not possible (since that triggers autodetection, which fails).

As a workaround, you can just edit `sketch.json` manually, of course.

Maybe it would be useful to add a `--fqbn` option to `board attach`, to allow specifying the FQBN in addition to the serial port and override autodetection? The interface might get a little messy, though. Maybe it would be better to then *also* add `--port` and stop support the `<port>|<FQBN>` non-option argument? i.e.:

```
Usage:
  arduino-cli board attach [sketchPath] [flags]

Examples:
  arduino-cli board attach --port serial:///dev/ttyACM0
  arduino-cli board attach --port serial:///dev/ttyACM0 HelloWorld
  arduino-cli board attach --fqbn arduino:samd:mkr1000
  arduino-cli board attach --fqbn arduino:samd:mkr1000 --port serial:///dev/ttyACM0
```

This would break compatibility, though...

- CLI version (output of `arduino-cli version`): Current git master 1855b531766056c9014df6f3e0aacad63d30ddc4
- OS and platform: Linux amd64

matthijskooijman, 2020-11-14:
A slightly related issue is that when specifying a port and doing autodetection, the "name" is set in `sketch.json`, but when specifying just the FQBN, this does not happen (even though, I think, the name could be looked up from the FQBN as well). See

https://github.com/arduino/arduino-cli/blob/1855b531766056c9014df6f3e0aacad63d30ddc4/commands/board/attach.go#L56-L59

and

https://github.com/arduino/arduino-cli/blob/1855b531766056c9014df6f3e0aacad63d30ddc4/commands/board/attach.go#L95-L99

ubidefeo, 2020-11-15:
@matthijskooijman 
I have noticed some odd behaviour in the `board attach` command, it definitely needs some love but it's a lower priority right now.

I have started reviewing this kind of workflows because in my opinion every time you need to pass a board model you should prefix it with the `-b | --fqbn` flag.
I think it makes up for greater consistency and better API

ubidefeo, 2020-12-28:
@matthijskooijman 
saw you linked this issue in another one
a reworked behaviour is part of the current sprint, but because of the holidays this sprint is gonna be longer than expected

cmaglie, 2022-11-21:
This problem has been resolved by the complete reworking of the board attach command in https://github.com/arduino/arduino-cli/pull/1930. The user preferences are now saved in the `sketch.yaml` file (that is the official Arduino CLI sketch project file) instead of the `sketch.json` file (that is the file used by the Arduino Web Editor).

closed 2022-11-21 by cmaglie