arduino/arduino-cli #1085: Add configuration option for `--warnings` (and maybe `--optimize-for-debug`)

matthijskooijman, 2020-11-25:
## Bug Report

### Current behavior

`arduino-cli compile` has a `--warnings` option to enable warnings for a single compilation run.

### Expected behavior

I would like to be able to set a default value for `--warnings` in my config file (i.e. to enable warnings on all compilations).

### Environment

- CLI version (output of `arduino-cli version`): recent git master
- OS and platform: Linux, x86_64.

ubidefeo, 2020-11-29:
@matthijskooijman 
would you really want `--optimize-for-debug` on as a default?
I think it's kinda counterproductive unless you add it to an extra config file that you point to during heavy development days, but seems like a stretch :D

maybe we could add a `-o` shortcut for that flag since it's available (cc @silvanocerza )

matthijskooijman, 2020-11-30:
>  would you really want --optimize-for-debug on as a default?

Possibly, yes. 99% (or something like that) of the uploads I do are during development, for testing, only a fraction is actually uploading a final product (and even then `-Og` is not really problematic). And regardless, it seems like something that would be easy to implement and the type of option that would make sense as a config directive along with enabling warnings, even if not a lot of people would end up using it.

> maybe we could add a -o shortcut for that flag since it's available (cc @silvanocerza )

That could also be a good idea, though I'd consider doing this *in addition to* adding a config option.

Maybe `-d` (for "debug") or `-g` (matching gcc) would be better, then?

ubidefeo, 2020-11-30:
>Possibly, yes. 99% (or something like that) of the uploads I do are during development

living on the edge :D

I'll push for a config setting, and I think I'd go for `-d` to keep it closer to _debug_ unless `-d` can be confusing for some reason

matthijskooijman, 2020-12-01:
Sounds good, thanks!

Just realized that this also means a `--no-optimize-for-debug` is needed, to reverse the setting from the config file (for the 1% where you *do* want to make a production build).

gilman88, 2023-06-06:
Hi just wanted to +1 having a warnings config setting since I prefer to see them. In particular this is because the permissive flag that is passed to the compiler is downgrading some errors and then they were being squashed due to the default warnings value.
An extra bonus would be if we could also have a flag/setting to prevent the permissive flag being sent to the compiler :)