arduino/arduino-cli #1086: Precompiled library, request to add a board specific sub-folder

landret, 2020-11-28:
Hi,
when using precompiled libraries, binaries are placed in folders according to their architecture. However, when building binaries for Uno and Nano, those binaries must be placed in the same folder called "atmega328p". So how is the good one picked by the Arduino IDE? Those binaries can be different in many ways, starting with the pin definition.

It would great to have sub-folders inside the architecture folder to differenciate boards (e.g. "atmega328p\uno", "atmega328p\nano").

facchinm, 2020-12-22:
Hi @landret ,
while I understand your request, the scope of precompiled libraries is to contain board-agnostic functions that should be linked at compile time.
So, my advise is to remove from the precompiled lib all board specific bits and provide them as APIs.

For example, if your library exposes function `foo()` that contains different code if compiled for UNO or Nano due to pin assignment, you should refactor it into something like `foo(pin1, pin2)` that is agnostic and then, from the sketch, make the right assignment.
If you are very time sensitive the best practice is to decouple it into two functions, one for UNO and one for Nano, and "hide" them in a higher level call that will contain the necessary ifdefs to choose the right one.

landret, 2020-12-26:
Hi @facchinm,
thanks for your advises, I will try to implement these.

closed 2021-05-11 by silvanocerza