arduino/arduino-cli #1092: Feature request: Expose sets of library files and directories needed for the compiler

ianfixes, 2020-12-01:
## Feature request

Part of unit test compilation in `arduino_ci` involves [determining which files and directories from a library should be presented to the compiler](https://github.com/Arduino-CI/arduino_ci/blob/master/lib/arduino_ci/cpp_library.rb#L503-L509).  I assume that `arduino-cli` already has logic to determine this same information, so I'm hopeful that it's as simple as exposing that behind a command or two.

This would eliminate any possible gaps in compiler behavior between the unit test compiler and the official one, especially if the behavior changes.

## Best Solution

All compilation information

```json
{
  "common_flags": [
    "-std=c++0x",
    "-O1"
  ],
  "dependencies": [
    "SomeDependency"
  ],
  "include_dirs": [
    "/path/to/library/src"
  ],
  "include_dirs_recursive": [
    "/path/to/library/src",
    "/path/to/SomeDependency/src"
  ],
  "source_files": [
    "/path/to/library/src/hello.cpp"
  ],
  "source_files_recursive": [
    "/path/to/library/src/hello.cpp",
    "/path/to/SomeDependency/src/hi.cpp"
  ]
}
```

### Next Best Solution

Authoritative information I can use to implement my own logic

```json
{
  "flat_layout": true,
  "base_directory": "/path/to/library/src"
}
```

umbynos, 2022-11-25:
Hi @ianfixes, we added a similar feature in the new versions of the CLI:

```
$ arduino-cli compile -b arduino:samd:mkrwifi1010 /home/umberto/Arduino/libraries/ArduinoOTA/examples/WiFi101_OTA
Sketch uses 26484 bytes (10%) of program storage space. Maximum is 262144 bytes.
Global variables use 3920 bytes (11%) of dynamic memory, leaving 28848 bytes for local variables. Maximum is 32768 bytes.

Used library Version Path                                                                        
SPI          1.0     /home/umberto/.arduino15/packages/arduino/hardware/samd/1.8.13/libraries/SPI
WiFiNINA     1.8.13  /home/umberto/Arduino/libraries/WiFiNINA                                    
ArduinoOTA   1.0.9   /home/umberto/Arduino/libraries/ArduinoOTA                                  

Used platform Version Path                                                          
arduino:samd  1.8.13  /home/umberto/.arduino15/packages/arduino/hardware/samd/1.8.13
```
It prints more info with the `--format json`

ianfixes, 2022-12-25:
I'm taking a look at this now as part of https://github.com/Arduino-CI/arduino_ci/pull/334

ianfixes, 2022-12-25:
I genuinely appreciate the response, and I'm grateful for the attention being paid to this issue.  That said, after reviewing the behavior in `0.29.0` I believe your https://github.com/arduino/arduino-cli/issues/1092#issuecomment-1327614240 misses 2 aspects of the original feature request:

1. Must return data _before_ the compiler is invoked, not after.  If the compilation fails, it doesn't look like `arduino-cli` will say anything about `provides_includes` in the JSON.
2. the output you posted shows the _extra libraries_ that were used in the sketch, but I'm interested in the local sources for all files in the library being compiled.  

The purpose of my request is to be able to ask a library which source files (according to the proper application of the library spec) must be passed to the compiler.

ianfixes, 2023-01-20:
Could we get the `status: waiting for information` label removed, or is there additional info needed?