arduino/arduino-cli #1095: Bug with recipe.size.regex in platform.txt

FrankBoesing, 2020-12-02:
If you try to set the regex in boards.txt and use it for
recipe.size.regex
it ends in a desaster :) 
example: recipe.size.regex = {build.flags.sizeregex}

A backslash gets doubled \ -> \\\ and the value gets surrounded by ""

Please fix this, as it can be useful if you have different boards with "inter-board" incompatible linkerfiles (in this regard)

FrankBoesing, 2021-02-14:
So, that was on Dec, 1st, last year.

Any news? Will that be fixed in the next decade?

FrankBoesing, 2021-03-03:
Ok, obviously not.

Closing here. Not interested anymore. Just leave that bug.

per1234, 2021-03-03:
Hi @FrankBoesing. Thank you very much for your report. If you don't mind, I'll leave this open. Even if it's not considered the top priority for the developers right now, I do think that it's important for bugs like this to be documented and tracked. If we don't keep track then there will be no chance of getting it fixed once the developers have some time to investigate.

FrankBoesing, 2021-03-03:
Just consider that unimportant things can be important for other users, and potentially even showstoppers.