arduino/arduino-cli #1100: on-by-default telemetry seeems sketchy to me

neinseg, 2020-12-05:
## Bug Report

### Current behavior

I just created a new project in a fresh ardunio-cli installation. I ran ``arduino-cli config init`` as the getting started guide recommended, and it created the ``arduino-cli.yaml`` as expected. When I had a look at that file, I noticed that it contained a ``telemetry`` setting that was on by default.

### Expected behavior

I have a strong opinion that I would like a setting like this to be off-by-default or explicit opt-in. At no point did arduino-cli or the Getting Started guide mention that there is telemetry in the first place, and I did not expect that kind of thing in this tool. From my user PoV, telemetry is an anti-feature.

A solution to this would be to ask the user "Would you like to enable telemetry (y/N)?" with default "no" during ``arduino-cli config init``.

### Environment

- CLI version (output of `arduino-cli version`):
arduino-cli Version: 0.13.0-arch Commit: 693a045eea420c29ca7027e668eee31bce37365d
- OS and platform:
Arch x86-64, up-to-date

### Additional context

Apart from just being not cool, having data collection on-by-default I think potentially exposes this project to a risk under GDPR liability. I am based in Germany, and had I not caught this by accident, arduino-cli would have disclosed at least the fact that I'm using Arduino along with my IP address to someone. IANAL but I think that is personal data, and I suppose that's not all ``telemetry`` it collects.

Have a nice day!

ianfixes, 2020-12-07:
As a maintainer of a software project that wraps `arduino-cli`, this is of interest to me as well.  When I auto-download this software, how do I also disable the telemetry in an automated fashion?

per1234, 2020-12-07:
@ianfixes you can find the documentation on how to configure Arduino CLI here:
https://arduino.github.io/arduino-cli/latest/configuration
Telemetry is controlled by the `telemetry.enabled` key in the configuration file or the `ARDUINO_TELEMETRY_ENABLED` environment variable.

ianfixes, 2020-12-07:
I didn't see `ARDUINO_TELEMETRY_ENABLED` listed in the URL you posted.  Am I correct in assuming that `ARDUINO_TELEMETRY_ENABLED=0` is the proper way to disable telemetry?

per1234, 2020-12-07:
>I didn't see ARDUINO_TELEMETRY_ENABLED listed in the URL you posted.

There's no point in documenting every environment variable name because they follow a standardized template which is documented:
https://arduino.github.io/arduino-cli/latest/configuration/#environment-variables
>The variable names start with ARDUINO, followed by the configuration key names, with each component separated by _. For example, the ARDUINO_DIRECTORIES_USER environment variable sets the directories.user configuration option.

>Am I correct in assuming that ARDUINO_TELEMETRY_ENABLED=0 is the proper way to disable telemetry?

That will work.

rsora, 2020-12-07:
Hi @neinseg

>  I think potentially exposes this project to a risk under GDPR liability

The `arduino-cli` currently **does not harvest any data** that is sent to Arduino. The telemetry implemented in the CLI is activated **only** in `daemon` mode and it collects statistics useful for observability purposes in a use-case where you integrate the CLI as a web service backend module.

@ianfixes
>  When I auto-download this software, how do I also disable the telemetry in an automated fashion?

If you are using the CLI as a command-line tool and not as a daemon in your project, no telemetry whatsoever is enabled. If you are running it in daemon mode, then @per1234 answer is the way to go.

You can look at the [documentation section](https://arduino.github.io/arduino-cli/latest/getting-started/#using-the-daemon-mode-and-the-grpc-interface) and https://github.com/arduino/arduino-cli/pull/573 which implements the functionality for more details.

Thanks for reaching out, if you have more questions on the topic, feel free to reply to this issue!

Happy Coding!

neinseg, 2020-12-07:
@rsora Oh, I get it! I got my understanding of that completely backwards!

So, "telemetry" here just enables a monitoring server integrated into the daemon, and it does not mean anything is ever sent to the greater internet. I read it like it was the other way around because [other companies](https://wiki.mozilla.org/Telemetry) call that "telemetry".

Thank you for the quick clarification. Have a nice day!

closed 2020-12-07 by neinseg