arduino/arduino-cli #1107: Arduino-cli doesn't generate hex files with bootloader if compiled size is > 64kiB

MCUdude, 2020-12-10:
Original issue [here](https://github.com/MCUdude/MegaCore/issues/147).

At least for AVR platforms, if the size of the compiled exceeds 65536 bytes, the [sketchname.ino].with_bootloader.hex file is not generated.  The reason why I'm stressing this is because [MightyCore](https://github.com/MCUdude/MightyCore) and [MegaCore](https://github.com/MCUdude/MegaCore) uses this file actively to prevent wiping the bootloader when uploading using a programmer

Here I'm compiling for Arduino MEGA 2560:

```
Sketch uses 65536 bytes (25%) of program storage space. Maximum is 253952 bytes.
```

Files in the `/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_435714/` folder:

[image]

Here is another example where the sketch is just two bytes larger:
```
Sketch uses 65538 bytes (25%) of program storage space. Maximum is 253952 bytes.
```
[image]

facchinm, 2020-12-10:
Was probably fixed by https://github.com/arduino/arduino-cli/pull/1017 . Did you try the latest prerelease version?

MCUdude, 2020-12-10:
> Was probably fixed by #1017 . Did you try the latest prerelease version?

Nope, I haven't. Will do now!

MCUdude, 2020-12-10:
Sorry, but the problem still occurs. This is the sketch I'm using to test this. It's an old Optiboot chaucer sketch I've trimmed down to be just the right size to trigger this. Just remove a few lines in the array to get the size below the "limit"

<details>

```c
// Text is an extract from The Canterbury Tales
// Full text at http://www.gutenberg.org/dirs/etext00/cbtls12.txt
#include <avr/pgmspace.h>

const char knightsTale[] PROGMEM= {
"                     THE KNIGHT'S TALE <1>\n"
"\n"
"\n"
"WHILOM*, as olde stories tellen us,                            *formerly\n"
"There was a duke that highte* Theseus.                   *was called <2>\n"
"Of Athens he was lord and governor,\n"
"And in his time such a conqueror\n"
"That greater was there none under the sun.\n"
"Full many a riche country had he won.\n"
"What with his wisdom and his chivalry,\n"
"He conquer'd all the regne of Feminie,<3>\n"
"That whilom was y-cleped Scythia;\n"
"And weddede the Queen Hippolyta\n"
"And brought her home with him to his country\n"
"With muchel* glory and great solemnity,                           *great\n"
"And eke her younge sister Emily,\n"
"And thus with vict'ry and with melody\n"
"Let I this worthy Duke to Athens ride,\n"
"And all his host, in armes him beside.\n"
"\n"
"And certes, if it n'ere* too long to hear,                     *were not\n"
"I would have told you fully the mannere,\n"
"How wonnen* was the regne of Feminie, <4>                           *won\n"
"By Theseus, and by his chivalry;\n"
"And of the greate battle for the nonce\n"
"Betwixt Athenes and the Amazons;\n"
"And how assieged was Hippolyta,\n"
"The faire hardy queen of Scythia;\n"
"And of the feast that was at her wedding\n"
"And of the tempest at her homecoming.\n"
"But all these things I must as now forbear.\n"
"I have, God wot, a large field to ear*                       *plough<5>;\n"
"And weake be the oxen in my plough;\n"
"The remnant of my tale is long enow.\n"
"I will not *letten eke none of this rout*.                *hinder any of\n"
"Let every fellow tell his tale about,                      this company*\n"
"And let see now who shall the supper win.\n"
"There *as I left*, I will again begin.                *where I left off*\n"
"\n"
"This Duke, of whom I make mentioun,\n"
"When he was come almost unto the town,\n"
"In all his weal, and in his moste pride,\n"
"He was ware, as he cast his eye aside,\n"
"Where that there kneeled in the highe way\n"
"A company of ladies, tway and tway,\n"
"Each after other, clad in clothes black:\n"
"But such a cry and such a woe they make,\n"
"That in this world n'is creature living,\n"
"That hearde such another waimenting*                      *lamenting <6>\n"
"And of this crying would they never stenten*,                    *desist\n"
"Till they the reines of his bridle henten*.                       *seize\n"
"\"What folk be ye that at mine homecoming\n"
"Perturben so my feaste with crying?\"\n"
"Quoth Theseus; \"Have ye so great envy\n"
"Of mine honour, that thus complain and cry?\n"
"Or who hath you misboden*, or offended?                         *wronged\n"
"Do telle me, if it may be amended;\n"
"And why that ye be clad thus all in black?\"\n"
"\n"
"The oldest lady of them all then spake,\n"
[... 1308 lines cut ...]
"And down he kneeleth, and with humble cheer*                  *demeanour\n"
};

void setup() {
  int i=0;
  uint8_t c;
  Serial.begin(115200);
  do {
    c = pgm_read_byte_far(pgm_get_far_address(knightsTale)+i);
    if (c) Serial.write(c);
    i++;
  } while (c);
  i = 0;
  do {
    c = pgm_read_byte_far(pgm_get_far_address(knightsTale2)+i);
    if (c) Serial.write(c);
    i++;
  } while (c);
}

void loop() {
}
```

</details>

facchinm, 2020-12-10:
Can you paste the verbose compilation output? And `arduino-cli --version` too. Thx

MCUdude, 2020-12-10:
> 

I don't have arduino-cli installed. All I have is the latest nightly IDE (1.8.14) whith arduino-cli bundled (I guess)

So when I run `arduino-cli --version`, I get 
```
-bash: arduino-cli: command not found
```

Where in the install can I find arduino-cli?

ubidefeo, 2020-12-10:
@MCUdude you are on the wrong repo
IDE 1.8.14 does not bundle the CLI, but since the mechanics are the same this might be an issue with CLI as well.
Would you be so kind to download our Alpha IDE and try that?
We'll be releasing a new version of this tomorrow, but it's worth giving it a shot
https://github.com/arduino/arduino-pro-ide#nightly-builds

MCUdude, 2020-12-11:
Sorry, I believed arduino-cli was bundled with the latest IDE. And the fact that I got search results for `with_bootloader` in the arduino-cli repo made me even more sure.

I just downloaded the Latest Pro IDE (nightly), and I can confirm that *with_bootloader.hex _is_ in fact generated when the sketch size is larger than 65536 bytes. Atleast on Windows. So I guess this bug relates to arduino-builder, not arduino-cli

facchinm, 2020-12-11:
`arduino-builder` is regularly rebuilt upon `arduino-cli`, so next java IDE will have the patch included :wink:

MCUdude, 2020-12-11:
> arduino-builder is regularly rebuilt upon arduino-cli, so next java IDE will have the patch included 😉

Awesome! Do you have an idea when the next IDE release (1.8.14) will be? The reason why I'm asking is that I'm not sure I should patch MightyCore and MegaCore to use the non *.with_bootloader.hex file as a temporary workaround or not. After all, the *.with_bootloader files are kinda useful, because you don't wipe the bootloader when you're uploading using a programmer. This is something that could be useful for the official Arduino AVR boards as well IMO 🙂

MCUdude, 2020-12-15:
@facchinm any hints? 😉

facchinm, 2020-12-15:
I just pushed the updated `arduino-builder` to master, so it should produce the hourly builds in a couple of hours. Would you mind testing it and reporting if it works fine? It would be a great speedup for release :wink:

MCUdude, 2020-12-15:
> I just pushed the updated arduino-builder to master, so it should produce the hourly builds in a couple of hours. Would you mind testing it and reporting if it works fine? It would be a great speedup for release 😉

Thank you! I'll give it a try tonight and report back

MCUdude, 2020-12-15:
@facchinm I just downloaded the latest nightly, and it does generate *.with_bootloader.hex now for sketches > 65536 bytes now. Horray!

MCUdude, 2020-12-16:
@facchinm another unrelated issue related to the nightly build that I'm not sure where to report is that the Avrdude version that's bundled with the nightly build is not up to date, and does not support the 32U4 based mEDBG programmer (found on the UNO Wifi Rev2). The fix is to install the ATmega megaAVR boards package, but why bundle an "old" Avrdude version with the IDE in the first place?

facchinm, 2020-12-16:
The nightly bundles just the classic AVR core and its dependencies, so it is expected for `avrdude` to just support the bundled targets. As you correctly stated, any additional or third party core that requires an updated version any tool will download and use it alongside the core files :slightly_smiling_face:

closed 2021-05-11 by silvanocerza