arduino/arduino-cli #1109: Undocumented (but minor) breaking change in 0.14

obra, 2020-12-15:
As part of our build system, we were running 'config init' twice on jobs in CI. Arguably this is a bug.

0.14 changed arduino-cli's behavior in such cases.

0.13 silently either ignored the issue or just overwrote the old config.

0.14 fails with the error `Config file already exists, use --overwrite to discard the existing one.`

I don't think this is a -bad- choice, but it was a little surprising to see jobs start failing. 

It'd be wonderful if stuff like this got into the docs, but I do understand if it's just too much of a burden during alpha.

silvanocerza, 2020-12-15:
Hi @obra, this is my fault I guess, I introduced this behaviour with #957 in September. 

We didn't have the breaking changes policy back then, during the latest release process I tried to document breaking changes introduced before the policy but must have missed some I guess.

I'll edit the `UPGRADING.md` file accordingly.

Thanks for the report!

closed 2020-12-15 by silvanocerza

silvanocerza, 2020-12-15:
Upgrading documentation has been updated accordingly and just published. I also updated the release changelog.
https://arduino.github.io/arduino-cli/0.14/UPGRADING/#change-behaviour-of-config-file-flag-with-config-commands

Thanks again for the issue, wouldn't have noticed if not for you. :)

obra, 2020-12-15:
No worries :)  Thanks!

PR #1110 (2020-12-15): [skip changelog] Add missing breaking change in UPGRADING.md
Adds a missed breaking change introduced before the breaking changes policy.
After merge I'll cherry pick this commit to `0.14.x` branch to update the documentation.

Fixes #1109.