arduino/arduino-cli #1111: Am I allowed to distribute arduino-cli via a PPA for Ubuntu?

Stig124, 2020-12-22:
I recently was thinking about distributing the arduino-cli binary file (and source package) by selfhosting a Debian repo, to permit easy installation, upgrade and remove of the package

Under GPLv3 I am allowed to "distribute" the software but I wanted to be sure that it was allowed

- The repo will use only original compiled binaries
- The source will be available on the same repo using `apt source`

ubidefeo, 2020-12-27:
hi @Stig124 

there shouldn't be any problem, but you'd have to track releases and update when a new one comes out.
Let me bring @rsora and @alranel into the chat, they are better at this than I'll ever be :)

Stig124, 2020-12-27:
> hi @Stig124
> 
> there shouldn't be any problem, but you'd have to track releases and update when a new one comes out.
> Let me bring @rsora and @alranel into the chat, they are better at this than I'll ever be :)

Great, 

I will track the releases automatically :

- As stated in [this page](https://arduino.github.io/arduino-cli/installation/) nightly build is done at 1AM GMT so with a cron script at ~2am GMT I would be able to get and upload the binary, and cloning/packaging of the source code would follow the same process
- And for stable releases I think I'll track them manually

Stig124, 2020-12-29:
Just to let you know, my repo is up and running for the stable branch in x86_64 (adding ARM will be very easy though)

Stig124, 2020-12-29:
I'll have a pull request to add my repo at the documentation done, feel free to ask anything you want about the repo

ubidefeo, 2020-12-29:
thank you for the heads up, @Stig124 :)

alranel, 2020-12-31:
@Stig124, GPLv3 allows you to do what you described, as long as you don't modify the arduino-cli sources or you provide the modified sources in case you do. :)

per1234, 2021-01-27:
Per https://github.com/arduino/arduino-cli/pull/1119#issuecomment-767800138, @Stig124 is no longer hosting the Arduino CLI APT repo.