arduino/arduino-cli #1113: Install library or core from GitHub branch or commit

woolseyj, 2020-12-26:
### Feature Request
In order to assist in development and testing, I would like to request the ability to install a core or library from a GitHub based branch or commit.  Something like the following is what I was thinking.

$ arduino-cli core install arduino:megaavr@branch:master
or
$ arduino-cli lib install WiFiNINA@commit:473d8b0

per1234, 2020-12-27:
`master` and `473d8b0` are both Git refs, so there's no need to give them different treatment.

>$ arduino-cli lib install WiFiNINA@commit:473d8b0

I think it would be unnecessarily limiting to use the library name, because that only allows installing libraries that are in the Library Manager index. Better would be to allow specifying a Git ref via the existing `arduino-cli lib install --git-url` command.

>$ arduino-cli core install arduino:megaavr@branch:master

It would be necessary to specify the repository URL. It it also necessary to specify the platform name because the repository doesn't contain this information. Another consideration is that not all platforms are in the root of the repository. Some (e.g., [MightyCore](https://github.com/MCUdude/MightyCore)) are structured with the platform under the architecture folder to allow easy manual installation. So to be universally useful this command would either need to allow the user to specify the installation source path relative to the root of the repository, or to automagically detect this path (e.g., search recusively for boards.txt)

alranel, 2020-12-31:
Another more modular approach could be to just accept a local path, and let the user clone the repo instead of embedding the git clone semantics in arduino-cli.

ubidefeo, 2020-12-31:
@alranel you can install libraries from local paths by supplying the path to a zip file with `--zip-path`.
remember that to enable installation of libraries from `git` URLs and zip files you need to enable `library.enable_unsafe_install: true` in your config file or as an ENV variable
```
library:
  enable_unsafe_install: true
```

alranel, 2021-01-08:
Yeah @ubidefeo, however that's not available for cores :)