arduino/arduino-cli #1114: arduino-cli uses wrong home directory on MacOS

plocher, 2020-12-26:
The directory search order for the config file is supposed to use $HOME...
```
jplocher@iMac-Pro ~ % echo $HOME
/Users/jplocher
```

note the "jplocher"...

Running the command from /Users/jplocher/:

```
jplocher@iMac-Pro ~ % arduino-cli --verbose config init
Error creating inventory dir: mkdir /Users/plocher: permission denied
ERRO[0000] Error creating inventory dir: mkdir /Users/plocher: permission denied 
Error writing inventory file: open /Users/plocher/Library/Arduino15/inventory.yaml: no such file or directory
ERRO[0000] Error writing inventory file: open /Users/plocher/Library/Arduino15/inventory.yaml: no such file or directory 
INFO[0000] Using config file: /Users/jplocher/Library/Arduino15/arduino-cli.yaml 
INFO[0000] arduino-cli version 0.14.0                   
INFO[0000] Writing config file to: /Users/plocher/Library/Arduino15 
Cannot create config file directory: mkdir /Users/plocher: permission denied
ERRO[0000] Cannot create config file directory: mkdir /Users/plocher: permission denied 
```

Where is the "/Users/plocher" (without the 'j') coming from?

the "plocher" directory name is not in Apple Preferences/Users, not in /etc/passwd, not in $ENV...
Could it be stored somewhere else that came from a different / previous installation elsewhere where that
home directory name had been used?

```
jplocher@iMac-Pro ~ % arduino-cli --verbose version
Error creating inventory dir: mkdir /Users/plocher: permission denied
ERRO[0000] Error creating inventory dir: mkdir /Users/plocher: permission denied 
Error writing inventory file: open /Users/plocher/Library/Arduino15/inventory.yaml: no such file or directory
ERRO[0000] Error writing inventory file: open /Users/plocher/Library/Arduino15/inventory.yaml: no such file or directory 
INFO[0000] Using config file: /Users/jplocher/Library/Arduino15/arduino-cli.yaml 
INFO[0000] arduino-cli version 0.14.0                   
arduino-cli alpha Version: 0.14.0 Commit: a86b21d99e2af9e0857da0ce4ab80baf1d3afb55 Date: 
```

plocher, 2020-12-26:
The startup code path in arduino-cli reads in ~/Library/Arduino15/arduino-cli.yaml

Root cause: This file is copied over by Apple's computer install/migration update process from a backup - even when the home directory names on the two computers are different.

Bug:  if the contents of this file are wrong, there is no indication in the error messages of what went wrong or how to  find, assess and fix it...

Suggested fix:
If the directories found in this file are invalid, catch the error and do something constructive about it.

Possibilities: 
- if the command is config init, ignore the info in this file and overwrite it with valid directory names.
- tell the user the name of this file (~/Library/Arduino15/arduino-cli.yaml) and highlight the incorrect directories found in it.
- ignore this file and write a new arduino-cli.yaml file in the current directory.
- ...etc...

plocher, 2020-12-26:
Work Around:
- Edit (or simply remove) the file ~/Library/Arduino15/arduino-cli.yaml

ubidefeo, 2020-12-27:
hi @plocher 

>Bug: if the contents of this file are wrong, there is no indication in the error messages of what went wrong or how to find, assess and fix it...

>Suggested fix:
If the directories found in this file are invalid, catch the error and do something constructive about it.

extremely good catch!
thank you for reporting it, we'll look into it :)

silvanocerza, 2021-01-11:
I've investigated a bit and we can't solve this because there's nothing to solve really.

The CLI is correctly failing in this case, if the paths are not valid is the user's responsability to update them to valid ones. 
If those paths don't exist the next time the CLI is run it tries to create them, this is done to gracefully handle first runs and whenever those paths are changed by the user. 

The CLI can't know wether there is a typo in the path or it's malformed in some other way so it just goes along with it and returns an error as soon as it happens.

ubidefeo, 2021-01-11:
@silvanocerza agreed

@plocher I hope you'll agree with this

plocher, 2021-01-11:
The bug isn't in what is done (i.e, fails), but in what isn't being done.
The current behavior is as expected, I agree.

The missing piece is the error message.

It needs to be clear that the path in a particular config file is invalid,
and call out both the path and the filename explicitly so that the user can
find and fix the problem.

The user never explicitly edited these files, and (probably?  certainly?)
doesn't even know they exist - or where they reside.
Thus, an error message needs to provide actionable context to enable a
manual fix.

The info on the web about arduino-cli is not at all clear on these files
names or exact locations.  I only found this level of detail by looking at
the bug reports here :-)

  -John

per1234, 2021-01-11:
>The info on the web about arduino-cli is not at all clear on these files
names or exact locations.

@plocher did you see this section of the Arduino CLI configuration documentation page?:
https://arduino.github.io/arduino-cli/dev/configuration/#configuration-file
To me, it seems clear about the filenames and locations, but I'm already so familiar with this subject matter that it's easy for me to miss deficiencies in the documentation. If you have any suggestions for improvement to that documentation it would be welcome.

plocher, 2021-01-11:
The "fix" I'm asking for is simply to add context to the initial error message output, which devotes 7 lines to the WRONG pathname and only one to the place to fix it :-)

```
jplocher@iMac-Pro ~ % arduino-cli --verbose config init
Error creating inventory dir: mkdir /Users/plocher: permission denied
ERRO[0000] Error creating inventory dir: mkdir /Users/plocher: permission denied 
Error writing inventory file: open /Users/plocher/Library/Arduino15/inventory.yaml: no such file or directory
ERRO[0000] Error writing inventory file: open /Users/plocher/Library/Arduino15/inventory.yaml: no such file or directory 
```
**INFO[0000] invalid directory path specified in directories.data:  /Users/plocher/Library/Arduino15**
INFO[0000] **Found in** config file: /Users/jplocher/Library/Arduino15/arduino-cli.yaml 
```
INFO[0000] arduino-cli version 0.14.0                   
INFO[0000] Writing config file to: /Users/plocher/Library/Arduino15 
Cannot create config file directory: mkdir /Users/plocher: permission denied
ERRO[0000] Cannot create config file directory: mkdir /Users/plocher: permission denied 
```

As an aside, one COULD make the call that` arduino-cli  config init` SHOULD overwrite the existing `arduino-cli.yaml` file in the user's Arduino15 system library directory WITHOUT first reading it, in which case the user would be able to fix this problem without having to invoke an editor ...

silvanocerza, 2021-01-12:
> The "fix" I'm asking for is simply to add context to the initial error message output, which devotes 7 lines to the WRONG pathname and only one to the place to fix it :-)

This makes sense. I can do something like this but I can't point where the path is read from without refactoring a big chunk of code.

```
Error: invalid path creating config dir: /home/<username>/.arduino15 error: mkdir /home/<username>: permission denied
```

> As an aside, one COULD make the call that arduino-cli config init SHOULD overwrite the existing arduino-cli.yaml file in the user's Arduino15 system library directory WITHOUT first reading it, in which case the user would be able to fix this problem without having to invoke an editor ...

Yeah, I wish for that too. The config file handling needs a big refactoring, that takes time to plan and execute it without breaking anything. It's not something that we can work on right now sadly.

By the way remember that you must use the `--overwrite` flag when calling `config init` if there's an already existing config file.