arduino/arduino-cli #1115: No libraries are found

logix5, 2020-12-27:
## Bug Report

When You try to search libraries, you always found any result. for example

C:\OttoBlockly\compilation\arduino>arduino-cli.exe lib search CCS811
Updating index: library_index.json downloaded
No libraries matching your search.

C:\OttoBlockly\compilation\arduino>arduino-cli.exe lib search debounce
Updating index: library_index.json downloaded
No libraries matching your search.

C:\OttoBlockly\compilation\arduino>arduino-cli.exe lib search softpwm
Updating index: library_index.json downloaded
No libraries matching your search.

### Current behavior

arduino-cli.exe lib search softpwm

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli.exe alpha Version: 0.14.0 Commit: a86b21d Date: 2020-12-10T14:00:57Z
- OS and platform: windows

### Additional context

ubidefeo, 2020-12-27:
hi @logix5 
I have had a similar issue the other day but now sure about the root of the problem.
can you try and launch the Arduino Java IDE, open the library manager and then try the CLI `lib search` command again?
I'm investigating if it can be an issue of broken (not-updated) library index and your support might come in handy.

I have been able to get results with the search terms you used

logix5, 2020-12-27:
Hi

I have tested this workaround and it is doesn't work me.  I have the
library manager open waiting to write the library to search. then i have
executed again the search and it didn't find any library:

C:\OttoBlockly\compilation\arduino>arduino-cli.exe lib search CCS811
Updating index: library_index.json downloaded
No libraries matching your search.

C:\OttoBlockly\compilation\arduino>arduino-cli.exe lib search debounced
Updating index: library_index.json downloaded
No libraries matching your search.

C:\OttoBlockly\compilation\arduino>arduino-cli.exe lib search adafruit
Updating index: library_index.json downloaded
No libraries matching your search.

Let me know you need more tests

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

ubidefeo, 2020-12-27:
@logix5 
does the IDE find the libraries correctly?

I suspect you might have a broken library index which can happen so you might need to force reload it.
If you head to the data folder specified in the config settings you should be able to delete `library_index.json` and `library_index.json.sig`

once you do that just run `arduino-cli update` and it should download everything fresh

logix5, 2020-12-27:
The IDE yes. Now i am using this way to install the libraries. Firstable
install using the arduino IDE and the copy this library en arduino-cli
folder

I will try this thing

I let you know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

logix5, 2020-12-27:
Thanks!!! It work!!

The issue can be closed. I don´t know why the file link was broken

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"