arduino/arduino-cli #1116: Attach doesnt work with esp8266

BernH4, 2020-12-27:
I installed the right core for my ESP8266 (esp8266:esp8266:nodemcuv2) and i can upload perfectly by specifying the port manually:
`arduino-cli upload -p /dev/ttyUSB0`

Altough i cant use the attach command with this board:
`arduino-cli board attach serial:///dev/ttyUSB0`
Output:
`Attach board error: no supported board found at serial:///dev/ttyUSB0`

I also tried it without `serial://` and by writing the port manually in the sketch.json file but this results in the error message:
`Error during Upload: uploading error: no upload port provided`
when running the command `arduino-cli upload`.

I can specify the board fine via attach.

### Environment

- CLI version: (arduino-cli alpha Version: 0.14.0-arch Commit: a86b21d99e2af9e0857da0ce4ab80baf1d3afb55 Date:
- OS and platform: ArchLinux running in a virtual Machine

per1234, 2020-12-27:
This is the expected behavior. The only way the board can be identified automatically is if it presents a unique VID/PID pair. This is the case with all the modern official Arduino boards, and some 3rd party boards, but the ESP8266 boards use general purpose USB to serial adapter chips (e.g., CH340) that use the VID/PID provided by the chip manufacturer. There is no way to know which board is on the other side of that general purpose chip.

Fortunately, board identification is merely a convenience feature and so this type of board is still perfectly usable with Arduino CLI.

More information: https://arduino.github.io/arduino-cli/latest/FAQ/#arduino-unomegaduemilanove-is-not-detected

matthijskooijman, 2020-12-27:
See also #1073, which requests being able to specify the FQBN *and* port to `attach`.

closed 2021-05-11 by silvanocerza