arduino/arduino-cli #1122: Archives files should not be embedded in archives

d-a-v, 2021-01-02:
## Bug Report

### Current behavior

In some case, library archives generated thanks to `dot_a_linkage=true` are put into `objs.a`,
resulting in an archive file embedded into another archive file.

Example in ESP8266Audio CI:
- 1st `.a` generated by [`dot_a_linkage=true`](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/library.properties) ([log in CI](https://github.com/earlephilhower/ESP8266Audio/pull/357/checks?check_run_id=1636167252#step:4:9897))
- 2nd .a generated by arduino-builder because of a too large number of generated files ([log in CI](https://github.com/earlephilhower/ESP8266Audio/pull/357/checks?check_run_id=1636167252#step:4:10373))
- [final error in CI](https://github.com/earlephilhower/ESP8266Audio/pull/357/checks?check_run_id=1636167252#step:4:10419)

probable source: https://github.com/arduino/arduino-cli/pull/961

### Expected behavior

Any of the two, or both ?
- refuse to add `.a` archive files in `AR` command line.
- skip `dot_a_linkage=true` libraries ?

### Environment

- CLI version (output of `arduino-cli version`):

arduino-builder shipped with arduino-ide-hourly-build

- OS and platform:

ubuntu / any

cmaglie, 2021-01-26:
Should be fixed by #1140, @d-a-v may you test with the latest nightly?

d-a-v, 2021-01-26:
Is Arduino-IDE-nightly including arduino-builder/cli nightly ?

per1234, 2021-01-27:
@d-a-v no. You can see the version of arduino-builder used in the nightly Arduino CLI build here:
https://github.com/arduino/Arduino/blob/master/build/build.xml#L102
```xml
  <property name="ARDUINO-BUILDER-VERSION" value="1.5.5" />
```
and the version of Arduino CLI used in that version of arduino-builder (confusing because they didn't push a 1.5.5 tag) here:
https://github.com/arduino/arduino-builder/blob/f959e8b154e695eaa3bfaac18f0310a854e228b8/go.mod#L7
```
	github.com/arduino/arduino-cli v0.0.0-20201210103408-bf7a3194bb63
```

cmaglie, 2021-01-27:
The CLI nightly is here: https://arduino.github.io/arduino-cli/latest/installation/#nightly-builds

As @per1234 said the Arduino IDE nightly does not get the CLI updates automatically, we have to manually update it.
@d-a-v if you need the patch in the Arduino IDE I can push it quickly.

d-a-v, 2021-01-27:
@per1234 @cmaglie Thanks !
Yes, that would allow @earlephilhower to try the fix in CI as described in OP.

cmaglie, 2021-01-27:
> Yes, that would allow @earlephilhower to try the fix in CI as described in OP.

Ok, the nightly of the Arduino IDE has been updated.

earlephilhower, 2021-01-28:
Thank you!

I can confirm the nightly build works successfully with the library that was seeing the issue before.

closed 2021-01-28 by per1234

cmaglie, 2021-01-28:
Thanks for the test!

Closing as fixed...