arduino/arduino-cli #1123: Compiler errors/warnings report from build directory instead of source directory for included files.

Marsfan, 2021-01-02:
## Bug Report

### Current behavior

```PowerShell
 .\arduino-cli.exe compile -v --config-file .\atp.yaml -b=SparkFun:apollo3:sfe_artemis_atp:svl_baud=921600,loader=option_svl D:\Desktop\test\test.ino
```

Errors and warnings produced by the compiler provide the path to the files in the temp directory instead of the actual file location. 
Example warning output. 

```PowerShell
C:\Users\Gabe\AppData\Local\Temp\arduino-sketch-D96E63BC1A299C47BBA385C4BB083A65\sketch\src\leptonSDK\FLIR_I2C.cpp: In function 'LEP_RESULT DEV_I2C_MasterWriteData(LEP_UINT16, LEP_UINT8, LEP_UINT16, LEP_UINT16*, LEP_UINT16, LEP_UINT16*, LEP_UINT16*)':
C:\Users\Gabe\AppData\Local\Temp\arduino-sketch-D96E63BC1A299C47BBA385C4BB083A65\sketch\src\leptonSDK\FLIR_I2C.cpp:254:59: error: call of overloaded 'write(LEP_UINT8*&, LEP_INT32&)' is ambiguous
     bytesActuallyWritten = Wire.write(txdata, bytesToWrite);
                                                           ^
```
### Expected behavior

Warnings for included files should instead point to the actual source file. 

Example Corrected Warning Output

```PowerShell
D:\Desktop\test\src\leptonSDK\FLIR_I2C.cpp: In function 'LEP_RESULT DEV_I2C_MasterWriteData(LEP_UINT16, LEP_UINT8, LEP_UINT16, LEP_UINT16*, LEP_UINT16, LEP_UINT16*, LEP_UINT16*)':
D:\Desktop\test\src\leptonSDK\FLIR_I2C.cpp:254:59: error: call of overloaded 'write(LEP_UINT8*&, LEP_INT32&)' is ambiguous
     bytesActuallyWritten = Wire.write(txdata, bytesToWrite);
                                                           ^
```

Note that in the corrected file output, the source file properly links to the actual user file that the error occurred in. 

### Environment

- CLI version (output of `arduino-cli version`): 0.14.0
- OS and platform: Windows 10 x64

### Additional context

This is absolutely necessary for properly implementing problem matching to allow for integrating `arduino-cli` into an IDE. With the outputs working the way that they currently do, clicking on errors will cause the file in the temporary directory to be opened, instead. This will cause the developer to edit a file that will simply be overwritten on the next build.

per1234, 2021-01-02:
Hi @Marsfan. Thanks for taking the time to submit an issue. We have a previous report for this: https://github.com/arduino/arduino-cli/issues/965 and even a proposed fix: https://github.com/arduino/arduino-cli/pull/707

Since it's better to concentrate all the discussion for each distinct topic in a single issue, I'll go ahead and close this as a duplicate.

Marsfan, 2021-01-02:
@per1234 No problem. I guess my searching skills failed me this time around.

closed 2021-05-11 by silvanocerza