arduino/arduino-cli #1127: Error: arduino-cli: no bottle available!

martsalz, 2021-01-06:
## Bug Report

### Current behavior

brew install arduino-cli
Error: arduino-cli: no bottle available!
You can try to install from source with:
  brew install --build-from-source arduino-cli
Please note building from source is unsupported. You will encounter build
failures with some formulae. If you experience any issues please create pull
requests instead of asking for help on Homebrew's GitHub, Twitter or any other
official channels.

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`):
- OS and platform:

MacBook Air M1 (Apple Silicon)

### Additional context

kumarpatel, 2021-02-13:
same issue

dansteingart, 2021-03-06:
bump, any update here?

ubidefeo, 2021-03-07:
hi everyone

Last week we fixed the Serial Discovery module to properly recognise boards on M1.
Once we release a fully M1 compliant CLI we'll update the bottle.

@martsalz ,  could you try `--build-from-source` again?

mik3y, 2021-03-16:
I'm seeing the following error with `--build-from-source` from an updated Homebrew on an M1 Mac:

```
# go.bug.st/serial
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:263:18: undefined: baudrateMap
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:268:23: undefined: baudrateMap
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:273:20: undefined: toTermiosSpeedType
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:274:20: undefined: toTermiosSpeedType
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:283:22: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:288:22: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:293:22: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:296:6: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:301:21: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:304:6: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:304:6: too many errors
```

Full log:

```
$ cat /Users/me/Library/Logs/Homebrew/arduino-cli/01.go
go
build
-trimpath
-o
/opt/homebrew/Cellar/arduino-cli/0.16.1/bin/arduino-cli
-ldflags
-s -w -X github.com/arduino/arduino-cli/version.versionString=0.16.1 -X github.com/arduino/arduino-cli/version.commit=76f55490438e6c19226c9e2d9934f7d088152648

# go.bug.st/serial
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:263:18: undefined: baudrateMap
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:268:23: undefined: baudrateMap
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:273:20: undefined: toTermiosSpeedType
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:274:20: undefined: toTermiosSpeedType
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:283:22: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:288:22: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:293:22: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:296:6: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:301:21: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:304:6: undefined: tcCMSPAR
/Users/me/Library/Caches/Homebrew/go_mod_cache/pkg/mod/go.bug.st/serial@v1.1.1/serial_unix.go:304:6: too many errors

HOMEBREW_VERSION: 3.0.5-66-g1e37fbd
ORIGIN: https://github.com/Homebrew/brew
HEAD: 1e37fbd362a9d91fe0f7fdf5202c05e5b83ce597
Last commit: 34 minutes ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 671f92993992851dbc607c8c5f7ac8483288fb55
Core tap last commit: 50 minutes ago
Core tap branch: master
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_MAKE_JOBS: 8
HOMEBREW_NO_BOOTSNAP: set
Homebrew Ruby: 2.6.3 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 12.0 build 1200
Git: 2.30.0 => /opt/homebrew/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 11.2.3-arm64
CLT: N/A
Xcode: 12.4
Rosetta 2: false

HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /opt/homebrew
CMAKE_INCLUDE_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/apache2:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/opt/homebrew/Library/Homebrew/os/mac/pkgconfig/11
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk
ACLOCAL_PATH: /opt/homebrew/share/aclocal
PATH: /opt/homebrew/Library/Homebrew/shims/mac/super:/opt/homebrew/opt/go/bin:/usr/bin:/bin:/usr/sbin:/sbin
```

timboldt, 2021-03-19:
Seems to work now after a `brew update`.

```
% brew install --build-from-source arduino-cli
==> Cloning https://github.com/arduino/arduino-cli.git
Cloning into '/Users/timboldt/Library/Caches/Homebrew/arduino-cli--git'...
==> Checking out tag 0.17.0
HEAD is now at ab58758 Improved error messages for .cpp/.h file in the sketch (#1224)
==> go build -ldflags -s -w -X github.com/arduino/arduino-cli/version.versionString=0.17.0 -X github.com/arduino/arduino-cli/version.commit=ab58758347e10f22e07154d520d0223612674cab
==> Caveats
zsh completions have been installed to:
  /opt/homebrew/share/zsh/site-functions
==> Summary
🍺  /opt/homebrew/Cellar/arduino-cli/0.17.0: 8 files, 19MB, built in 29 seconds
```

martsalz, 2021-03-19:
@ubidefeo , it works!

mik3y, 2021-03-22:
Confirmed as well - thanks!

silvanocerza, 2021-05-10:
Closing this since it seems like it has been solved.