arduino/arduino-cli #1131: Arduino not generating hex files with bootloader

ruggedvideo, 2021-01-12:
## Bug Report
This is similar to the issue in #1107 https://github.com/arduino/arduino-cli/issues/1107
### Current behavior

We have a basic project in Arduino IDE using ATTiny13.  For the first batch of boards, we programmed in-circuit using the Sparkfun AVR Pocket Programmer.  First we burn the bootloader, then we upload the sketch.  

Now, we are trying to export a hex file with bootloader, so we can have chips pre-programmed at the Microchip factory.  When we use the Export compiled Binary option, we expected two hex files to be generated, one "with bootloader" and one without; however, only one hex file is created, presumably without bootloader because the filename does not include the text "bootloader".   

One difference between this and the issue described in #1107 is that our sketch size is only 266 bytes.  We're not sure if we are doing something wrong or if this is a bug, but after some web searching it seems like probably a bug.

[image]

Below is the debug output from the Export operation.  

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Seth\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Seth\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Seth\Documents\Arduino\libraries -fqbn=MicroCore:avr:attiny13:clock=9M6,BOD=2v7,lto=Os_flto -ide-version=10809 -build-path C:\Users\Seth\AppData\Local\Temp\arduino_build_167138 -warnings=none -build-cache C:\Users\Seth\AppData\Local\Temp\arduino_cache_526188 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Seth\Downloads\RV-10222S_RevA\RV-10222S_RevA.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Seth\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Seth\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Seth\Documents\Arduino\libraries -fqbn=MicroCore:avr:attiny13:clock=9M6,BOD=2v7,lto=Os_flto -ide-version=10809 -build-path C:\Users\Seth\AppData\Local\Temp\arduino_build_167138 -warnings=none -build-cache C:\Users\Seth\AppData\Local\Temp\arduino_cache_526188 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Seth\Downloads\RV-10222S_RevA\RV-10222S_RevA.ino
Using board 'attiny13' from platform in folder: C:\Users\Seth\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.4
Using core 'microcore' from platform in folder: C:\Users\Seth\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.4
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore" "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13" "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\sketch\\RV-10222S_RevA.ino.cpp" -o nul -D COMPILER_LTO
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore" "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13" "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\sketch\\RV-10222S_RevA.ino.cpp" -o "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\preproc\\ctags_target_for_gcc_minus_e.cpp" -D COMPILER_LTO
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore" "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13" "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\sketch\\RV-10222S_RevA.ino.cpp" -o "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\sketch\\RV-10222S_RevA.ino.cpp.o" -D COMPILER_LTO
Compiling libraries...
Compiling core...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore" "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13" "C:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore\\millis.S" -o "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\core\\millis.S.o" -D COMPILER_LTO
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore" "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13" "C:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore\\WInterrupts.c" -o "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\core\\WInterrupts.c.o" -D COMPILER_LTO
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore" "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13" "C:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore\\wiring.c" -o "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\core\\wiring.c.o" -D COMPILER_LTO
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore" "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13" "C:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore\\wiring_digital.c" -o "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\core\\wiring_digital.c.o" -D COMPILER_LTO
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore" "-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13" "C:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore\\wirin [... cut ...] 0222S_RevA.ino.hex"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138/RV-10222S_RevA.ino.elf"
Sketch uses 266 bytes (25%) of program storage space. Maximum is 1024 bytes.
Global variables use 9 bytes (14%) of dynamic memory, leaving 55 bytes for local variables. Maximum is 64 bytes.

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`):
- OS and platform:

### Additional context

rsora, 2021-01-12:
Hi @ruggedvideo, 

Can you please paste here the complete `arduino-cli` command (or commands) that you are using? 

Thanks for reaching out!

ruggedvideo, 2021-01-12:
Sorry, I think I logged this in the wrong place... we are using the IDE.

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine
-hardware C:\Program Files (x86)\Arduino\hardware -hardware
C:\Users\Seth\AppData\Local\Arduino15\packages -tools C:\Program Files
(x86)\Arduino\tools-builder -tools C:\Program Files
(x86)\Arduino\hardware\tools\avr -tools
C:\Users\Seth\AppData\Local\Arduino15\packages -built-in-libraries
C:\Program Files (x86)\Arduino\libraries -libraries
C:\Users\Seth\Documents\Arduino\libraries
-fqbn=MicroCore:avr:attiny13:clock=9M6,BOD=2v7,lto=Os_flto
-ide-version=10809 -build-path
C:\Users\Seth\AppData\Local\Temp\arduino_build_167138 -warnings=none
-build-cache C:\Users\Seth\AppData\Local\Temp\arduino_cache_526188
-prefs=build.warn_data_percentage=75 -verbose
C:\Users\Seth\Downloads\RV-10222S_RevA\RV-10222S_RevA.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine
-hardware C:\Program Files (x86)\Arduino\hardware -hardware
C:\Users\Seth\AppData\Local\Arduino15\packages -tools C:\Program Files
(x86)\Arduino\tools-builder -tools C:\Program Files
(x86)\Arduino\hardware\tools\avr -tools
C:\Users\Seth\AppData\Local\Arduino15\packages -built-in-libraries
C:\Program Files (x86)\Arduino\libraries -libraries
C:\Users\Seth\Documents\Arduino\libraries
-fqbn=MicroCore:avr:attiny13:clock=9M6,BOD=2v7,lto=Os_flto
-ide-version=10809 -build-path
C:\Users\Seth\AppData\Local\Temp\arduino_build_167138 -warnings=none
-build-cache C:\Users\Seth\AppData\Local\Temp\arduino_cache_526188
-prefs=build.warn_data_percentage=75 -verbose
C:\Users\Seth\Downloads\RV-10222S_RevA\RV-10222S_RevA.ino
Using board 'attiny13' from platform in folder:
C:\Users\Seth\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.4
Using core 'microcore' from platform in folder:
C:\Users\Seth\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.4
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g
-Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections
-fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=9600000L
-DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g
"-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore"
"-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13"
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\sketch\\RV-10222S_RevA.ino.cpp"
-o nul -D COMPILER_LTO
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g
-Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections
-fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=9600000L
-DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g
"-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore"
"-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13"
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\sketch\\RV-10222S_RevA.ino.cpp"
-o
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\preproc\\ctags_target_for_gcc_minus_e.cpp"
-D COMPILER_LTO
"C:\\Program Files
(x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u
--language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns
--line-directives
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g
-Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections
-fno-threadsafe-statics -MMD -mmcu=attiny13a -DF_CPU=9600000L
-DARDUINO=10809 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g
"-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\cores\\microcore"
"-IC:\\Users\\Seth\\AppData\\Local\\Arduino15\\packages\\MicroCore\\hardware\\avr\\1.0.4\\variants\\attiny13"
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\sketch\\RV-10222S_RevA.ino.cpp"
-o
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138\\sketch\\RV-10222S_RevA.ino.cpp.o"
-D COMPILER_LTO
[... 179 lines cut ...]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objcopy" -O
ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings
--change-section-lma .eeprom=0
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138/RV-10222S_RevA.ino.elf"
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138/RV-10222S_RevA.ino.eep"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objcopy" -O
ihex -R .eeprom
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138/RV-10222S_RevA.ino.elf"
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138/RV-10222S_RevA.ino.hex"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A
"C:\\Users\\Seth\\AppData\\Local\\Temp\\arduino_build_167138/RV-10222S_RevA.ino.elf"
Sketch uses 266 bytes (25%) of program storage space. Maximum is 1024 bytes.
Global variables use 9 bytes (14%) of dynamic memory, leaving 55 bytes for
local variables. Maximum is 64 bytes.

Seth Gudmundson
Rugged Video LLC
ruggedvid.com
414-399-3370 office
*NEW ADDRESS*
*W66N205 Commerce Court*
*Cedarburg, WI 53012*

Visit us at HAI Heli-Expo Booth 11428!
March 22nd to 25th, New Orleans, LA

per1234, 2021-01-12:
Due to the limited memory of the ATtiny13, MicroCore does not use a bootloader. The reason you still might do a **Tools > Burn Bootloader** when using MicroCore is because it sets the configuration fuses on the ATtiny13 according to the custom Tools menu selections (BOD, EEPROM, Clock). You can see here that the bootloader command doesn't actually flash anything to the chip:
https://github.com/MCUdude/MicroCore/blob/master/avr/platform.txt#L142

So don't worry about the lack of the *_bootloader.hex file. Just provide the manufacturer with the normal .hex file and make sure to tell them which fuse settings you require.

ruggedvideo, 2021-01-12:
Thanks!

Seth Gudmundson
Rugged Video LLC
ruggedvid.com
414-399-3370 office
*NEW ADDRESS*
*W66N205 Commerce Court*
*Cedarburg, WI 53012*

Visit us at HAI Heli-Expo Booth 11428!
March 22nd to 25th, New Orleans, LA

closed 2021-05-11 by silvanocerza