arduino/arduino-cli #1132: Unable to compile sketch

gadhagod, 2021-01-13:
## Bug Report

### Current behavior

When running: 

    arduino-cli compile --fqbn arduino:avr:uno sketch

I receive:

    Error during build: fork/exec /bin/avr-g++: no such file or directory

### Expected behavior

I expect it to compile.

### Environment

arduino-cli alpha Version: 0.14.0 Commit: a86b21d Date: 2020-12-10T13:59:22Z
- OS and platform: Raspberry Pi Zero, Linux

---

I installed on my mac, and everything went smooth, but when trying on a RPi, I get that error. I installed using:

    curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh

Please let me know if there is any more information that I could give to help fix this problem with the Arduino CLI.

ubidefeo, 2021-01-13:
@gadhagod 
in the past we had issues with some tools not being available or not correctly installed on the Zero because of its limited RAM.
I think this issue has been resolved.

Have you installed the correct platforms?
please try the following

```
arduino-cli cache clean
arduino-cli core install arduino:avr
``
and then try the compilation again

let me know
u.

gadhagod, 2021-01-13:
I receive the same result. Does anyone know what `/bin/avr-g++` should be?

ubidefeo, 2021-01-14:
@gadhagod  can you try run

`arduino-cli core uninstall arduino:avr -v`

then

`arduino-cli core install arduino:avr -v`
 and then try again?

is `sketch` a new sketch you have created?

closed 2021-05-11 by silvanocerza