arduino/arduino-cli #1142: Error: Error during build: exit status -1073741515

prorishi, 2021-01-16:
## Bug Report

### Current behavior

I have an Amica NodeMCU board and trying to upload the following sketch. I have installed the esp8266:esp8266 core and the board is also listed in `arduino-cli board listall` command
when iam trying to compile it the error occurs `Error during build: exit status -1073741515`
but when i tried to compile it for arduino nano, it was compiled and then uploaded succesfully

`arduino-cli.exe compile --fqbn esp8266:esp8266:nodemcuv2:xtal=160,vt=flash,exception=enabled,ssl=all,eesz=4M,led=16,ip=hb2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 bl`

bl is a simple blink sketch:
```
void setup() {
        Serial.begin(115200);
        pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
        digitalWrite(LED_BUILTIN, HIGH);
        Serial.println("Led ON");
        delay(800);
        digitalWrite(LED_BUILTIN, LOW);
        Serial.println("Led OFF");
        delay(800);
}
```

### Expected behavior

i had expected the sketch to be compiled  without any errors 

### Environment

- CLI version (output of `arduino-cli version`):
          arduino-cli.exe alpha Version: 0.14.0 Commit: a86b21d Date: 2020-12-10T14:00:57Z
- OS and platform:
          Windows 7 Ultimate Service Pack 1, 32-bit OS

ubidefeo, 2021-01-17:
@prorishi 
unfortunately I don't have a Win 7 machine to test.
could you try 

`arduino-cli core uninstall esp8266:esp8266`
and then
`arduino-cli core install esp8266:esp8266`

I have successfully compiled on win10 and Mac OS, so your issue might even lie in your system path.
If it fails compiling please launch the command adding `-v` at the end of the line and paste the whole output in a comment

prorishi, 2021-01-18:
## @ubidefeo Thanks for your concern

Today, I tried again with the same command an everything went smooth
uploading was also successful

`arduino-cli.exe compile --fqbn esp8266:esp8266:nodemcuv2:xtal=160,vt=flash,exception=enabled,ssl=all,eesz=4M,led=16,ip=hb2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 bl`
```Executable segment sizes:
IROM   : 266356          - code in flash         (default or ICACHE_FLASH_ATTR)
IRAM   : 27264   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...)
DATA   : 1252  )         - initialized variables (global, static) in RAM/HEAP
RODATA : 824   ) / 81920 - constants             (global, static) in RAM/HEAP
BSS    : 25008 )         - zeroed variables      (global, static) in RAM/HEAP
Sketch uses 295696 bytes (28%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27084 bytes (33%) of dynamic memory, leaving 54836 bytes for local variables. Maximum is 81920 bytes.
```

Thank you so much.

closed 2021-01-18 by ubidefeo