arduino/arduino-cli #1149: Please add .cc and .cxx to OTHER_ALLOWED_EXTENSIONS

raybellis, 2021-01-22:
When using third party C++ code not originally targeting the Arduino it is not uncommon to find those source files with a ".cc" or ".cxx" extension.

These are currently unusable since the builder doesn't copy those files from `src/*` into the temporary build tree.

Please add those to the list of recognised source code extensions for C++ code.

per1234, 2021-01-22:
Just for the sake of linking related feature requests together, I'll mention https://github.com/arduino/arduino-builder/issues/268 here.

raybellis, 2021-01-22:
Oops, not sure how this ended up in `arduino-cli` instead of the core UI repository!

per1234, 2021-01-22:
Hi @raybellis. I transferred it to this repository because this is where the code that would need to be changed to implement your request lives. To be specific, here:
https://github.com/arduino/arduino-cli/blob/900654f45c6761c51568420d79f4f9be0e785fff/arduino/globals/globals.go#L27-L43

The builder system has gradually been moved out of the Arduino IDE's Java codebase, first into [arduino-builder](https://github.com/arduino/arduino-builder), and now to its final home here in Arduino CLI. From here, it can be used by all the official (and 3rd party as well if they like) Arduino development software, including Arduino IDE, Arduino Web Editor, Arduino Pro IDE, and of course Arduino CLI itself.

Sag06, 2021-12-18:
Today dev code. Arduino IDE 1.8.13.  .ipp files are not valid though

exit status 1
led_pattern.re.ipp: No such file or directory
web_asyncwebprint.ipp: No such file or directory

I could compile changing file extensions to .h and include references

chenzhuoyu, 2023-01-07:
Any updates? @raybellis

closed 2023-08-23 by cmaglie