arduino/arduino-cli #1152: Ability to customize name of binary output.

draco2003, 2021-01-26:
## Bug Report

I use system for OTA Updates which requires binaries with a certain naming schema. 
The previous version I had been using this worked perfectly. Changed machines and installed the latest to find that option isn't available. 

Previous arduino-cli version had an --output flag that allowed me to have something like this in my build script. 

`--output ../firmware/"$BUILDNAME"-"$BUILDVERSION".bin ` 

It would be great to have this functionality back so that we don't have to jump through hoops to rename the file after building it. 

### Environment

- CLI version (output of `arduino-cli version`): 0.14.0
- OS and platform: Raspberrypi x64

silvanocerza, 2021-01-26:
Which version were you using before that the `--output` flag? We're talking about the `compile` command right? Just want to be sure.

draco2003, 2021-01-29:
Definitely appreciate the desire to confirm. Don't want to waste any time. 

Had to do some digging, but looks like this commit is the one that removed the "output" option and switched to the "output-dir" option. Would love the output option back, or an outputfilename option or something similar.

https://github.com/arduino/arduino-cli/commit/ebc28e1bb15818b24b180221a898a50c8be306c7#diff-4cfe4c705335bdee380ac833a37b2b10b1ba7fbccb521eb78e596e33225d2c12

Which leads to this PR that mentions the output deprecation:
https://github.com/arduino/arduino-cli/pull/687

Which lead me to search more closed issues and finally found this one:
https://github.com/arduino/arduino-cli/issues/983
Specifically the answer of not adding that back:
https://github.com/arduino/arduino-cli/issues/983#issuecomment-699892728

Will go the "build a rename/cleanup script" route to do it after builds, but want to officially give getting that feature back a +1 where ever we can register those requests :)

closed 2021-05-11 by silvanocerza