arduino/arduino-cli #1154: mixed code libraries (again?)

gizero, 2021-01-26:
I can't a successful build for a sketch that depends on https://github.com/BoschSensortec/BSEC-Arduino-library. I'm completely new to Arduino environments even though I do have a reasonable background in build systems in general. I know BSEC library has already been matter of discussion in the past because it provides mixed sources and prebuilt artefacts which must be both used when linking the final binary. I found in particular #611 which seems to have fixed a likely related issue. IIUC, BSEC is even used as a test case for arduino-cli these days (https://github.com/arduino/arduino-cli/blob/master/test/test_compile.py#L207). However, I'm using a more recent version of the library ([v1.6.1480](https://github.com/BoschSensortec/BSEC-Arduino-library/releases/tag/v1.6.1480)).

Moreover, I'm not sure to understand #611, when it says "https://github.com/BoschSensortec/BSEC-Arduino-library (works after removing precompiled=true directive in library.properties) -> mixed (different code and precompiled)". Do I need to "fix" the library: to me `precompiled=true` seems the correct option at least according to https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format.

Any advice welcome and sorry for the noise just in the case this is trivial and already documented somewhere, but I'm been struggling a few hours already and I do feel stuck with it.

### Current behavior

The error happens at the link stage: here is the relevant build sub-command output. If required I can provide more details and the full log.
```
/Users/gizero/work/fluidware/fcub/msp-firmware/var/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc -nostdlib -L/Users/gizero/work/fluidware/fcub/msp-firmware/var/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib -L/Users/gizero/work/fluidware/fcub/msp-firmware/var/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group -Wl,--whole-archive /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/sketch/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/FS/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SD/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SPI/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/Wire/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/WiFi/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/aead/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/codec/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/ec/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/hash/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/int/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/kdf/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/mac/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/rand/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/rsa/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/ssl/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/symcipher/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/SSLClient/bearssl/src/x509/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/BSEC_Software_Library/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/PMS_Library/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/MiCS6814-I2C/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/U8g2/objs.a /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/U8g2/clib/objs.a -Wl,--no-whole-archive /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/core/core.a -lgcc -lesp32 -lphy -lesp_http_client -lmbedtls -lrtc -lesp_http_server -lbtdm_app -lspiffs -lbootloader_support -lmdns -lnvs_flash -lfatfs -lpp -lnet80211 -ljsmn -lface_detection -llibsodium -lvfs -ldl_lib -llog -lfreertos -lcxx -lsmartconfig_ack -lxtensa-debug-module -lheap -ltcpip_adapter -lmqtt -lulp -lfd -lfb_gfx -lnghttp -lprotocomm -lsmartconfig -lm -lethernet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_supplicant -lmesh -lesp_https_ota -lwpa2 -lexpat -llwip -lwear_levelling -lapp_update -ldriver -lbt -lespnow -lcoap -lasio -lnewlib -lconsole -lapp_trace -lesp32-camera -lhal -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr -lstdc++ -L/Users/gizero/work/fluidware/fcub/msp-firmware/var/libraries/BSEC_Software_Library/src/esp32 -lalgobsec -lalgobsec -Wl,--end-group -Wl,-EL -o /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/msp-firmware.ino.elf
/Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/BSEC_Software_Library/objs.a: member /Users/gizero/work/fluidware/fcub/msp-firmware/var/build/libraries/BSEC_Software_Library/objs.a(BSEC_Software_Library.a) in archive is not an object
collect2: error: ld returned 1 exit status
```

### Expected behavior

The same sketch builds fine with Arduino IDE 1.8.13.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: 0.14.0 Commit: a86b21d99e2af9e0857da0ce4ab80baf1d3afb55 Date:
- OS and platform: macOS 10.14.6

matthijskooijman, 2021-01-26:
I think this is a duplicate of #1122.

cmaglie, 2021-01-26:
yes it's a duplicate and should be already fixed by https://github.com/arduino/arduino-cli/pull/1140
@gizero may you try the latest nightly build?

gizero, 2021-01-26:
@cmaglie latest nightly does indeed runs the build to completion! Thanks for pointing me to the right direction: hope to see this fix released soon. Will help in setting up a reliable CI with arduino-cli. Closing this one.

closed 2021-05-11 by silvanocerza