arduino/arduino-cli #1155: Not downloading board package due to non existing signature

Craft4Cube, 2021-01-28:
## Bug Report

### Current behavior

```bash
# arduino-cli --config-file arduino-cli.yaml -v board listall                                                                                                                                                                                                                                                           ✔  2161  10:28:25
INFO[0000] Using config file: /tmp/test/arduino-cli.yaml 
INFO[0000] arduino-cli version 0.14.1                   
INFO[0000] Checking signature                            index=data/package_index.json signatureFile=data/package_index.json.sig trusted=true
INFO[0000] Checking signature                            error="opening signature file: open data/package_STM32duino_index.json.sig: no such file or directory" index=data/package_STM32duino_index.json signatureFile=data/package_STM32duino_index.json.sig
INFO[0000] Checking if CLI is Bundled into the IDE      
INFO[0000] Loading hardware from: data/packages         
INFO[0000] Loading package builtin from: /private/tmp/test/data/packages/builtin 
INFO[0000] Checking existence of 'tools' path: /private/tmp/test/data/packages/builtin/tools 
INFO[0000] Loading tools from dir: /private/tmp/test/data/packages/builtin/tools 
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.1.0"
INFO[0000] Adding libraries dir                          dir=/tmp/test/libraries location=user
Board Name FQBN 
```

Even though the Board Package url is given, arduino-cli will not install it.

### Expected behavior

Download and install the package, like the Arduino IDE does. The link given works fine in the IDE.

### Environment

- CLI version (output of `arduino-cli version`):
`arduino-cli alpha Version: 0.14.1 Commit: 0284b57a459ae1ee9d2adeacdb08b68daa5c4160 Date:`
- OS and platform:
`Darwin mbp.local 19.6.0 Darwin Kernel Version 19.6.0: Tue Nov 10 00:10:30 PST 2020; root:xnu-6153.141.10~1/RELEASE_X86_64 x86_64`

Craft4Cube, 2021-01-28:
arduino-cli.yaml:
```yaml
board_manager:
  additional_urls: ["http://dan.drown.org/stm32duino/package_STM32duino_index.json"]
daemon:
  port: "50051"
directories:
  data: ./data
  downloads: ./staging
  user: ./
library:
  enable_unsafe_install: true
logging:
  file: ""
  format: text
  level: info
sketch:
  always_export_binaries: true
telemetry:
  addr: :9090
  enabled: false
```

per1234, 2021-01-28:
Hi @Craft4Cube. I think that error message is acting as a "red herring". It is only a problem when the package index file is hosted on arduino.cc and shouldn't cause any problems for package index files on other hosts.

Did you remember to install the STM32 platform? You can do that by running one of the following commands:
```
arduino-cli core install stm32duino:STM32F1
```
or
```
arduino-cli core install stm32duino:STM32F4
```
Depending on which boards you are using (the http://dan.drown.org/stm32duino/package_STM32duino_index.json offers two different platforms).

Craft4Cube, 2021-01-28:
Welp, that fixed it. Thanks for the hint!

closed 2021-05-11 by silvanocerza