arduino/arduino-cli #1168: Please update "host": descripion in index.json specification

ArminJo, 2021-02-04:
Hello,
in the [current specification](https://arduino.github.io/arduino-cli/package_index_json-specification/) I see:
```
    Linux 64-bit (x86_64-linux-gnu),
    Linux 32-bit (i686-linux-gnu),
    Windows (i686-mingw32),
    Mac (i386-apple-darwin11)
```
In the Arduino package_index.json there are also aarch64-linux-gnu + arm-linux-gnueabihf.
And what confuses me, is that there are new entries x86_64-**pc**-linux-gnu / i686-**pc**-linux-gnu.
What is the difference to x86_64-linux-gnu / i686-linux-gnu?

Thanks
Armin

per1234, 2021-02-04:
Hi @ArminJo. The list you see in the Package Index Specification is only some examples, not a comprehensive list of all supported values. I agree that the supported values being undocumented is not so great.

You can see the full list here:
https://github.com/arduino/arduino-cli/blob/a9b0c9df7f8dca8cac87b5337c583b4510c5fc56/arduino/cores/tools.go#L143-L190

>there are also aarch64-linux-gnu + arm-linux-gnueabihf

`aarch64-linux-gnu` is 64 bit Linux on ARM architecture host.
`arm-linux-gnueabihf` is 32 bit Linux on ARM architecture host.

>What is the difference to x86_64-linux-gnu / i686-linux-gnu?

You can see that in the code above the regular expresssions provide a good deal of flexibility in the host values.

This sort of variation certainly has caused me some confusion, since I tend to try to reverse engineer the platform system by studying the existing platforms, rather than diving into the actual code. This would lead me to assume there was some significance between these different host values, but as far as Boards Manager is concerned they are identical.

ArminJo, 2021-02-04:
T H A N K S 🥇

closed 2021-05-11 by silvanocerza