arduino/arduino-cli #1173: gRPC sketch content list should stop at first level

ubidefeo, 2021-02-05:
## Bug Report
Issue is related to https://github.com/arduino/arduino-pro-ide/issues/416 
I deduced that the call for file listing issued by the IDE over gRPC gets the CLI to recursively go through the tree hence returning every file.
This creates the issue linked above.
For how the Sketch is structured, there should be no recursion, and only files at the main Sketch level should be returned.

### Environment

- CLI version (output of `arduino-cli version`): 0.15.x
- OS and platform: All

### Additional context
https://github.com/arduino/arduino-pro-ide/issues/416

silvanocerza, 2021-02-09:
I investigated a bit and we probably make changes both in the CLI and the IDE to fix this.

The data returned by the gRPC interface is that returned by `LoadSketch`:
https://github.com/arduino/arduino-cli/blob/0425c95acc227742fd9705ebd60c338e82dc9359/commands/instances.go#L731-L753

If we try to load a sketch structured like this:

```
TestLoadSketchFolder
├── doc.txt
├── header.h
├── old.pde
├── other.ino
├── s_file.S
├── src
│   ├── dont_load_me.ino
│   └── helper.h
└── TestLoadSketchFolder.ino
```

the resulting data in `LoadSketchResp` would be:

* `MainFile: TestLoadSketchFolder.ino`
* `LocationPath: <absolute_path_to_sketch_folder>`
* `OtherSketchFiles: [old.pde, other.ino]`
* `AdditionalFiles: [header.h, s_file.S, src/helper.h]`

The IDE loads `MainFile`, `OtherSketchFiles` and `AdditionalFiles`.

To make it work the way we expect it we must make the CLI append to `OtherSketchFiles` all the supported files from the sketch root folder and put the rest in `AdditionalFiles`.

The IDE must not open `AdditionalFiles` as tabs.

This was introduced with PR #926

Would like some feedback from @kittaakos and @ubidefeo before going forward with this.

kittaakos, 2021-02-10:
@silvanocerza, I am not sure if I can advise here but your proposal looks good to me. Adjusting the IDE code is not a problem at all, we can filter the additional files from the sketch. I checked how the [Marlin](https://github.com/MarlinFirmware/Marlin) project behaves in the Java IDE:
[image]

If the CLI can return with the same resultset, we are good here. The only question I would have is how to open the other files if opening the other files is a requirement at all.

silvanocerza, 2021-02-10:
We'll go with the Java IDE behaviour for now. We'll see in the future what to do I guess.

silvanocerza, 2021-02-10:
I tried to implent what I suggested above but sadly it can't be done. The CLI uses those properties in the compilation process, changing them would break lots of parts of the legacy package. 

Think I'll just a `RootFolderFiles` that will contain all the files in the sketch root. Nonetheless we'll need to think of a better way of handling files in the IDE.

kittaakos, 2021-02-10:
> Think I'll just a `RootFolderFiles` that will contain all the files in the sketch root.

This can be done on the IDE side too. No need for a change now, the IDE can wait until the CLI team figures out what is the correctt way to handle it.

PR #1182 (2021-02-11): Add RootFolderFiles to gRPC LoadSketchResp
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for breaking changes)

* **What ki [... cut]