arduino/arduino-cli #1190: [Question] How can I make the arduino-cli be embedding the ardunio:avr platform in the core from the first?

JeongJun-Lee, 2021-02-19:
Hello,

I want to release this arduino-cli for offline users using Arduino Uno hardware.
So how can I make the arduino-cli be embedding the ardunio:avr platform in the core from the first?  
Because the offline users don't have any Internet connection, they can't install the additional core ardunio:avr platform for compiling later by online. 

Thanks,
JJ

ubidefeo, 2021-02-19:
hi @JeongJun-Lee 
You could try and place a custom `arduino-cli.yaml` file where your CLI is installed manually.
Here you can decide the path for everything.
Of course you will have to manually copy the core package.

If you look at how your `Arduino15` folder is structured after you have the CLI running and installing on your machine for the first time, you could copy the same structure or customise the CLI even further using the config file

closed 2021-02-20 by per1234

JeongJun-Lee, 2021-02-21:
@ubidefeo Thanks a lot for your reply. It works! I have to write a command with the configuration's location specified by the --config-file command line flag. So I am going to release this CLI including the` Arduino15` directory. But the `Arduino15` directory size is almost 340MB. Why is it so big for `Arduoino Uno` only? It's two times bigger than Arduino IDE. How can I reduce the size of release? Thanks.

per1234, 2021-02-21:
Hi @JeongJun-Lee. The toolchain and platform archive files for the Arduino AVR Boards package installation are cached in the folder location set via the [`directories.downloads` configuration key](https://arduino.github.io/arduino-cli/dev/configuration/#configuration-keys). That is set to the `staging` subfolder of `Arduino15` by default.

The cached archive files are useful in the event someone wants to uninstall and then reinstall a package, because they avoid the need to download the files from the Internet again. However, they do greatly increase the size of the Arduino15 folder and are only a convenience. So you can safely delete that folder.

This issue tracker is only to be used to report bugs or feature requests. Support requests should be made to the dedicated board for Arduino CLI on the Arduino Forum:
https://forum.arduino.cc/index.php?board=145.0
I closely monitor that board and am always happy to help out. In the event there is a question I'm unable to answer, I can request assistance from the Arduino developers. So it would be better to continue any further discussion over there. In the event those discussions reveal defects or required features in Arduino CLI, then it will be great to open focused issues here for them.

JeongJun-Lee, 2021-02-21:
@per1234  Thank you for your reply. Let's keep on in Arduino Forum as you mentioned.  I made a new [post]<https://forum.arduino.cc/index.php?topic=728982>