arduino/arduino-cli #1192: Please add a way to see all the configurable options for a board

jockm, 2021-02-22:
Right now I have two projects going, one using MightyCore for the ATMega644p, and the other using ATTinyCore for the ATtiny85, neither specify the clock speed in the same way.  Every time I use a new board variant, I have to go to the github page for it, find `boards.txt` and then scan the file to see everything I might want to customize

I would like a way to see everything that is configurable.  I imagine the command might look something like:

` arduino-cli board listconfig {fqbn}`

per1234, 2021-02-22:
Your wish is granted. The command is `arduino-cli board details`
https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_board_details/
```
$ arduino-cli board details -b MightyCore:avr:644
Board name:            ATmega644
FQBN:                  MightyCore:avr:644
Board version:         2.0.8

Package name:          MightyCore
Package maintainer:    MCUdude
Package URL:           https://mcudude.github.io/MightyCore/package_MCUdude_MightyCore_index.json
Package website:       https://github.com/MCUdude/MightyCore
Package online help:   http://forum.arduino.cc/index.php?topic=379427

Platform name:         MightyCore
Platform category:     Contributed
Platform architecture: avr
Platform URL:          https://MCUdude.github.io/MightyCore/MightyCore-2.0.8.tar.bz2
Platform file name:    MightyCore-2.0.8.tar.bz2
Platform size (bytes): 625615
Platform checksum:     SHA-256:3227ed90e1c5163883f1e93d659a57828201cd07a3447fe564338da457bcb1ee

Option:                Bootloader                                                                                    bootloader
                       Yes (UART0)                                           ✔                                       bootloader=uart0
                       Yes (UART1)                                                                                   bootloader=uart1
                       No bootloader                                                                                 bootloader=no_bootloader
Option:                EEPROM                                                                                        eeprom
                       EEPROM retained                                       ✔                                       eeprom=keep
                       EEPROM not retained                                                                           eeprom=erase
Option:                Pinout                                                                                        pinout
                       Standard pinout                                       ✔                                       pinout=standard
                       Bobuino pinout                                                                                pinout=bobuino
                       Sanguino pinout                                                                               pinout=sanguino
Option:                Variant                                                                                       variant
                       644P / 644PA                                          ✔                                       variant=modelP
                       644 / 644A                                                                                    variant=modelA
Option:                BOD                                                                                           BOD
                       BOD 2.7V                                              ✔                                       BOD=2v7
                       BOD 4.3V                                                                                      BOD=4v3
                       BOD 1.8V                                                                                      BOD=1v8
                       BOD disabled                                                                                  BOD=disabled
Option:                Compiler LTO                                                                                  LTO
                       Disabled                                              ✔                                       LTO=Os
                       Enabled                                                                                       LTO=Os_flto
Option:                Clock                                                                                         clock
                       External 16 MHz                                       ✔                                       clock=16MHz_external
                       External 20 MHz                                                                               clock=20MHz_external
                       External 18.4320 MHz                                                                          clock=18_432MHz_external
                       External 14.7456 MHz                                                                          clock=14_7456MHz_external
                       External 12 MHz                                                                               clock=12MHz_external
                       External 11.0592 MHz                                                                          clock=11_0592MHz_external
                       External 8 MHz                                                                                clock=8MHz_external
                       External 7.3728 MHz                                                                           clock=7_3728MHz_external
                       External 4 MHz                                                                                clock=4MHz_external
                       External 3.6864 MHz                                                                           clock=3_6864MHz_external
                       External 2 MHz                                                                                clock=2MHz_external
[... cut ...]
                       avrispmkii                                            AVRISP mkII (MightyCore)
                       jtag3                                                 Atmel JTAGICE3 (JTAG mode) (MightyCore)
                       atmel_ice                                             Atmel-ICE (AVR) (MightyCore)
                       avrisp                                                AVR ISP (MightyCore)
                       usbasp_slow                                           USBasp slow (MightyCore)
```

jockm, 2021-02-22:
Awesome! Thank you for pointing that out, I have no idea how I missed it

closed 2021-05-11 by silvanocerza