arduino/arduino-cli #1196: panic when installing a lib from ZIP and directories.data does not exist

kittaakos, 2021-02-24:
## Bug Report

I got a panic when I wanted to install a lib from ZIP.

### Current behavior

```
akos.kitta@Akoss-MacBook-Pro arduino-cli_nightly-20210224_macOS_64bit % rm -rf ~/Library/Arduino15                                       
akos.kitta@Akoss-MacBook-Pro arduino-cli_nightly-20210224_macOS_64bit % rm -rf ~/Documents/Arduino                                       
akos.kitta@Akoss-MacBook-Pro arduino-cli_nightly-20210224_macOS_64bit % ./arduino-cli version --format json                              
{
  "Application": "arduino-cli",
  "VersionString": "nightly-20210224",
  "Commit": "4c9ce83e",
  "Status": "alpha",
  "Date": "2021-02-24T01:32:19Z"
}
akos.kitta@Akoss-MacBook-Pro arduino-cli_nightly-20210224_macOS_64bit % ./arduino-cli lib install --zip-path ~/Desktop/Tlc5940-0.15.0.zip --verbose
INFO[0000] Config file not found, using default values  
INFO[0000] arduino-cli version nightly-20210224         
INFO[0000] Checking if CLI is Bundled into the IDE      
INFO[0000] Loading hardware from: /Users/akos.kitta/Library/Arduino15/packages 
INFO[0000] Adding libraries dir                          dir=/Users/akos.kitta/Documents/Arduino/libraries location=user
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                                                                                                          
INFO[0000] Installing tool                               Tool="builtin:ctags@5.8-arduino11"
Installing builtin:ctags@5.8-arduino11...
INFO[0000] Tool installed                                Tool="builtin:ctags@5.8-arduino11"
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.1.0...
builtin:serial-discovery@1.1.0 downloaded                                                                                                                       
INFO[0000] Installing tool                               Tool="builtin:serial-discovery@1.1.0"
Installing builtin:serial-discovery@1.1.0...
INFO[0001] Tool installed                                Tool="builtin:serial-discovery@1.1.0"
builtin:serial-discovery@1.1.0 installed
INFO[0001] Loading hardware from: /Users/akos.kitta/Library/Arduino15/packages 
INFO[0001] Loading package builtin from: /Users/akos.kitta/Library/Arduino15/packages/builtin 
INFO[0001] Checking existence of 'tools' path: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0001] Loading tools from dir: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0001] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0001] Loaded tool                                   tool="builtin:serial-discovery@1.1.0"
WARN[0001] There were errors loading the library index, trying again... 
INFO[0001] Updating libraries index                     
Updating index: library_index.json downloaded                                                                                                                   
INFO[0001] Loading hardware from: /Users/akos.kitta/Library/Arduino15/packages 
INFO[0001] Loading package builtin from: /Users/akos.kitta/Library/Arduino15/packages/builtin 
INFO[0001] Checking existence of 'tools' path: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0001] Loading tools from dir: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0001] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0001] Loaded tool                                   tool="builtin:serial-discovery@1.1.0"
INFO[0001] Adding libraries dir                          dir=/Users/akos.kitta/Documents/Arduino/libraries location=user
INFO[0001] Loading hardware from: /Users/akos.kitta/Library/Arduino15/packages 
INFO[0001] Loading package builtin from: /Users/akos.kitta/Library/Arduino15/packages/builtin 
INFO[0001] Checking existence of 'tools' path: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0001] Loading tools from dir: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0001] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0001] Loaded tool                                   tool="builtin:serial-discovery@1.1.0"
INFO[0001] Adding libraries dir                          dir=/Users/akos.kitta/Documents/Arduino/libraries location=user
INFO[0002] URL: https://downloads.arduino.cc/packages/package_index.json 
INFO[0002] Updating index                                url="https://downloads.arduino.cc/packages/package_index.json"
Updating index: package_index.json downloaded                                                                                                                   
Updating index: package_index.json.sig downloaded                                                                                                               
INFO[0002] Checking signature                            error="opening signature file: open /var/folders/k3/d2fkvv1j16v3_rz93k7f74180000gn/T/811020046.sig: no such file or directory" index=/var/folders/k3/d2fkvv1j16v3_rz93k7f74180000gn/T/811020046 signatureFile=/var/folders/k3/d2fkvv1j16v3_rz93k7f74180000gn/T/811020046.sig
INFO[0002] Checking signature                            index=/Users/akos.kitta/Library/Arduino15/package_index.json signatureFile=/Users/akos.kitta/Library/Arduino15/package_index.json.sig trusted=true
INFO[0002] Loading hardware from: /Users/akos.kitta/Library/Arduino15/packages 
INFO[0002] Loading package builtin from: /Users/akos.kitta/Library/Arduino15/packages/builtin 
INFO[0002] Checking existence of 'tools' path: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0002] Loading tools from dir: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0002] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0002] Loaded tool                                   tool="builtin:serial-discovery@1.1.0"
INFO[0002] Adding libraries dir                          dir=/Users/akos.kitta/Documents/Arduino/libraries location=user
INFO[0002] Checking signature                            index=/Users/akos.kitta/Library/Arduino15/package_index.json signatureFile=/Users/akos.kitta/Library/Arduino15/package_index.json.sig trusted=true
INFO[0002] Loading hardware from: /Users/akos.kitta/Library/Arduino15/packages 
INFO[0002] Loading package builtin from: /Users/akos.kitta/Library/Arduino15/packages/builtin 
INFO[0002] Checking existence of 'tools' path: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0002] Loading tools from dir: /Users/akos.kitta/Library/Arduino15/packages/builtin/tools 
INFO[0002] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0002] Loaded tool                                   tool="builtin:serial-discovery@1.1.0"
INFO[0002] Adding libraries dir                          dir=/Users/akos.kitta/Documents/Arduino/libraries location=user
panic: runtime error: index out of range [1] with length 1

goroutine 1 [running]:
github.com/arduino/arduino-cli/cli/lib.runInstallCommand(0xc0016cb340, 0xc0016c6b70, 0x1, 0x3)
	/home/build/cli/lib/install.go:69 +0x1245
github.com/spf13/cobra.(*Command).execute(0xc0016cb340, 0xc0016c6b40, 0x3, 0x3, 0xc0016cb340, 0xc0016c6b40)
	/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:846 +0x29d
github.com/spf13/cobra.(*Command).ExecuteC(0xc0000ab8c0, 0x0, 0xc00164d9e0, 0x0)
	/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:950 +0x349
github.com/spf13/cobra.(*Command).Execute(...)
	/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:887
main.main()
	/home/build/main.go:31 +0x85
akos.kitta@Akoss-MacBook-Pro arduino-cli_nightly-20210224_macOS_64bit % 
```

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`): `nightly-20210224`
- OS and platform: macOS 11.2.1

### Additional context

kittaakos, 2021-02-24:
It's affecting the nightlies only.

Ansari-Tech, 2021-09-28:
Any update on this? still getting this problem on **alpha Version: nightly-20210928 Commit: 4708cde Date: 2021-09-28T01:27:37Z**

Need to be on nightly because of #1448 which seems to have not been built to release yet.

PR #1479 (2021-09-29): [skip changelog] Fix lib install crash on nightlies
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for breaking change [... cut]
closed 2021-09-29 by silvanocerza