arduino/arduino-cli #1206: All links for "More info" on Arduino boards on Boards Manager point to Arduino website front-page

elsatch, 2021-03-02:
As I was reviewing the new Arduino IDE 2.0.0-beta.3, I was trying to install the supporting library for Arduino Nano 33 BLE Sense. 

I opened the Boards Manager from the tool menu and looked for the board. Looking for additional information to see if I should install Arduino mbed-enabled board or Arduino nRF52 board package, I discovered that **all the links for "More info" on Arduino boards are pointing to the front page of the Arduino website, instead of the library information**. This makes the "More info" button for libraries quite useless.

[image]

P.D If you look for Sense, or Arduino Nano 33 BLE Sense in board manager nothing appears in the search results. This might be a separate bug too.

per1234, 2021-03-02:
This is the same behavior as the classic Arduino IDE. The "More info" link is defined by the [package index](https://arduino.github.io/arduino-cli/dev/package_index_json-specification/)'s  `packages[].websiteURL`. That is indeed set to http://www.arduino.cc/ for all the official Arduino boards platforms:
https://downloads.arduino.cc/packages/package_index.json

The classic Arduino IDE has a separate "Online help" link, defined by `packages[].platforms[].help.online`
[image]
It's this link that is is missing from Arduino IDE 2.x's Boards Manager.

It looks like the gRPC interface's [`PlatformSearchResp`](https://arduino.github.io/arduino-cli/dev/rpc/commands/#platformsearchresp) doesn't currently provide this information, so I think work will first need to be done to Arduino CLI, which is why I have transferred this issue here rather than the (`arduino/arduino-ide`)[https://github.com/arduino/arduino-ide] repo.

elsatch, 2021-03-03:
Thanks for the update. As I was reading again the issue, I just realized I was on the board manager and not the library manager. So most of my comments about library info were a bit misguided, as they should point to the board family information.

closed 2022-12-19 by MatteoPologruto