arduino/arduino-cli #1208: arduino-cli compile - Error during build: no such file /../../bin/avr-g++

ThorpeJosh, 2021-03-04:
## Bug Report

### Current behavior
```bash
$ arduino-cli compile -v --verify --upload -p /dev/ttyUSB0 -b arduino:avr:uno path_to_script/GateSketch
Using board 'uno' from platform in folder: /root/.arduino15/packages/arduino/hardware/avr/1.8.3
Using core 'arduino' from platform in folder: /root/.arduino15/packages/arduino/hardware/avr/1.8.3
Detecting libraries used...
/root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/root/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/root/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/standard /tmp/arduino-sketch-2C02EEF97571C887912CB7C968D29AB0/sketch/GateSketch.ino.cpp -o /dev/null

Error during build: fork/exec /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: no such file or directory
```
### Expected behavior

I could compile code and upload using version 0.14.0 but now on 0.16.1 I get this error
### Environment

arduino-cli alpha Version: 0.16.1 Commit: 76f55490 Date: 2021-02-25T10:43:25Z
- OS and platform:
  Raspberry Pi Model B, Raspbian Buster.
  

### Additional context
I run this arduino compilation in a docker container but I also tried on the host and same issue.

The file it claims does not exist, does exist.
```bash
ls -lah /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
-rwxr-xr-x 2 root root 743K Mar  4 09:24 /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++

matthijskooijman, 2021-03-04:
> The file it claims does not exist, does exist.

I think (memory is a bit vauge) this same (and confusing) error message also occurs when there is a mismatch in architecture (i.e. 32bit vs 64bit) for a binary, or when a library needed by the binary is not available (or not in the right architecture).

Could you run these two commands and show us the output?

```
file /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
ldd /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
```

ThorpeJosh, 2021-03-04:
Hi,

```bash
# uname -a
Linux 8b28ba6b2802 5.10.14+ #1401 Mon Feb 8 14:19:47 GMT 2021 armv6l GNU/Linux
# file /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
/root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, stripped
# ldd /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
not a dynamic executable
```

I have a working instance of arduino-cli and the output is different only on the ldd command (obviously).

cheers,

matthijskooijman, 2021-03-04:
Output looks ok, except "not a dynamic executable" seems unexpected, especially since the `file` command says it *is* a dynamic executable. But the command is ARMv5 and your kernel is ARMv6, so that should work, also everything 32-bit AFAICS.

One other command to try:

```
strace -f /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
```

This might need you to install the `strace` package first. This command should show all the syscalls done by an executable, I hope it shows which (library) file is triggering the `No such file or directory` exactly.

ThorpeJosh, 2021-03-04:
```
# strace -f /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
execve("/root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++", ["/root/.arduino15/packages/arduin"...], 0xbeb1be14 /* 8 vars */) = -1 ENOENT (No such file or directory)
strace: exec: No such file or directory
+++ exited with 1 +++```

ThorpeJosh, 2021-03-04:
I think it has truncated the file in the output?

matthijskooijman, 2021-03-04:
Hm, this strace does not really help, since it's the exec call that fails, so I can't see what happens behind the scenes for this exec call. I don't have any more ideas just now, maybe @facchinm has some experience with the Rpi build?

ThorpeJosh, 2021-03-11:
Update: Compliation works on older RPi (armv6l) when arduino-sli is installed on the host OS. It fails only when arduino-cli is installed and run in a docker container.

However, on a newer RPi (armv7l) arduino-cli compiles on host OS and in a docker container without issues.

What is the difference between a docker container on an armv6 and armv7 host?

umbynos, 2022-11-30:
Hi @ThorpeJosh, sorry for the delay in the response...
Could you please try with the latest version of the CLI?

If the problem still exists, could you please the docker configuration you have used?

nuno-silva, 2023-01-18:
Hi! Not sure if this is actually the same exact problem, but I was getting this error on a Raspberry Pi 3 B (ARMv7):

```sh
$ arduino-cli compile --fqbn esp32:esp32:esp32 

Used platform Version Path                                                        
esp32:esp32   2.0.6   /home/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.6

Error during build: fork/exec /home/arduino/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++: no such file or directory
```

The file does exist:
```sh
$ file /home/arduino/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++
/home/arduino/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=d80d264c42b092c58ad16f1f1f454d13664e109a, stripped
```

But I am unable to run it (hence the confusing message from arduino-cli):

```sh
$ /home/arduino/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++
-bash: /home/arduino/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++: No such file or directory

$ strace /home/arduino/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++
execve("/home/arduino/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++", ["/home/arduino/.arduino15/package"...], 0x7eb82070 /* 32 vars */) = -1 ENOENT (No such file or directory)
strace: exec: No such file or directory
+++ exited with 1 +++
```

---

Some more system info:

```sh
$ arduino-cli version
arduino-cli  Version: 0.29.0 Commit: 76251df9 Date: 2022-11-17T09:21:45Z

$ arduino-cli core list
ID          Installed Latest Name 
esp32:esp32 2.0.6     2.0.6  esp32

$ uname -a
Linux pi3 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l ARMv7 Processor rev 4 (v7l) BCM2835 GNU/Linux
$ cat /proc/cpuinfo | head -n 10
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 70.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

```
Upon further investigation, I noticed I'm missing the `/lib/ld-linux.so.3` interpreter needed by the binary.

Maybe not the ideal solution, but I fixed this by symlinking the missing interpreter:
```sh
$ sudo ln -s  /lib/ld-linux-armhf.so.3  /lib/ld-linux.so.3

$ /home/arduino/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++       
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
```

umbynos, 2023-01-19:
@nuno-silva does this happen also with the arduino:avr core? Like reported in the original issue?

nuno-silva, 2023-01-20:
hey @umbynos ! I don't have access to the Raspberry pi now but I'll give it a try next week and report back.

nuno-silva, 2023-02-02:
hey @umbynos, sorry for the delay. **I deleted the symlink** and tested using arduino:avr core.

Using

- arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7
- arduino:avrdude@6.3.0-arduino17
- arduino:arduinoOTA@1.3.0
- arduino:avr@1.8.6

An empty sketch compiles with no problem:

```sh

~/Arduino/Example $ arduino-cli compile --fqbn arduino:avr:uno
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

Used platform Version Path                                                        
arduino:avr   1.8.6   /home/arduino/.arduino15/packages/arduino/hardware/avr/1.8.6

```

Note avr-gcc in this case seems to be using the correct interpreter:

```
$ file .arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc          
.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, stripped
```

orhun, 2023-02-23:
Got this on `Raspberry Pi 3 Model B Rev 1.2` running `Alpine Linux v3.17 aarch64` and it was due to Arduino tools using `glibc`:

```
$ file /home/orhun/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++

/home/orhun/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, for GNU/Linux 3.7.0, BuildID[sha1]=fc933c0814da959514cf03a2e47f58905aadb382, stripped
```

```
$ ldd /home/orhun/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
	/lib/ld-linux-aarch64.so.1 (0x7fb6ff7000)
	libc.so.6 => /lib/ld-linux-aarch64.so.1 (0x7fb6ff7000)
Error loading shared library ld-linux-aarch64.so.1: No such file or directory (needed by /home/orhun/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++)
```

It is solved by installing the `glibc` compatibility package:

```
$ apk add gcompat
```

Just my 2 cents.

ChillarAnand, 2023-09-20:
Same issue is happening on ARMv7 RPi as well. https://github.com/arduino/arduino-cli/issues/2328

DansDesigns, 2025-01-16:
Having same issue on a NanoPi Neo armv7l running Armbian 24.11.1, trying to compile for esp32:

`bin/xtensa-esp32-elf-g++: no such file or directory`

file is in the folder specified, I have tried many version of esp32 core and all give the same error on both root and normal user

anayrat, 2025-11-05:
> Having same issue on a NanoPi Neo armv7l running Armbian 24.11.1, trying to compile for esp32:
> 
> `bin/xtensa-esp32-elf-g++: no such file or directory`
> 
> file is in the folder specified, I have tried many version of esp32 core and all give the same error on both root and normal user

Same here for ESP32 on Raspbian Trixie and RPi 2

```
Used platform Version Path
esp32:esp32   3.3.3   /home/anayrat/.arduino15/packages/esp32/hardware/esp32/3.3.3
Error during build: fork/exec /home/anayrat/.arduino15/packages/esp32/tools/esp-x32/2507/bin/xtensa-esp32-elf-g++: no such file or directory

file /home/anayrat/.arduino15/packages/esp32/tools/esp-x32/2507/bin/xtensa-esp32-elf-g++
/home/anayrat/.arduino15/packages/esp32/tools/esp-x32/2507/bin/xtensa-esp32-elf-g++: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=53bef39eacc90f3a7c5007e7245bc2f8d321d35e, stripped

ls /lib/ld-linux.so.3
ls: cannot access '/lib/ld-linux.so.3': No such file or directory
```