arduino/arduino-cli #1220: Invalid sketch.json is silently ignored

matthijskooijman, 2021-03-11:
## Bug Report

### Current behavior

```
matthijs@grubby:~/test$ tree
.
├── sketch.json
└── test.ino

0 directories, 2 files
matthijs@grubby:~/test$ cat sketch.json 
{ foo
matthijs@grubby:~/test$ arduino-cli compile -v

Error during build: no FQBN provided

```
When an invalid sketch.json is present, it is silently ignored.

### Expected behavior

I would expect an error about sketch.json containing invalid JSON and no other messages.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: 0.0.0-git Commit:  Date: 
- Not sure why `arduino-cli version` says nothing useful,  but this is a locally compiled, slightly modified version based on bf364db76bd763bc5326750090005042d7058506 from Feb 26.
- OS and platform: Linux / Ubuntu Focal / amd64

per1234, 2021-03-11:
>Not sure why arduino-cli version says nothing useful

This happens when you use `go build` instead of `task build`. Task build generates the data used by `arduino-cli version`:
https://github.com/arduino/arduino-cli/blob/436277f6738c27c2ec5fe657b32827b06ad28c5d/Taskfile.yml#L217-L223

matthijskooijman, 2021-03-11:
> This happens when you use go build instead of task build. Task build generates the data used by arduino-cli version:

Makes sense, thanks!

umbynos, 2022-11-30:
We dropped support for `sketch.json` in favor of `sketch.yaml` in #1930