arduino/arduino-cli #1225: CLI returning multiple FBQN when only one board connected for STM32

toogooda, 2021-03-16:
**Describe the bug**
Com port selection for IDE is not working because CLI is returning multiple rows t, It should look like this when you connect an STM32 board:
[image]

Instead you get this:
[image]

What is also strange is that if you select any of them they all get selected and the same when deselecting
[image]

With just one bluepill connected CLI returns multiple rows:
```
c:\Program Files (x86)\Arduino_CLI>arduino-cli board list --format json
[
  {
    "address": "COM5",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)",
    "boards": [
      {
        "name": "Maple Mini",
        "FQBN": "stm32duino:STM32F1:mapleMini",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Maple (RET6)",
        "FQBN": "stm32duino:STM32F1:mapleRET6",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103C series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103C",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103T series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103T",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103Z series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103Z",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Microduino Core STM32 to Flash",
        "FQBN": "stm32duino:STM32F1:microduino32_flash",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103R series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103R",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "HYTiny STM32F103TB",
        "FQBN": "stm32duino:STM32F1:hytiny-stm32f103t",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Maple (Rev 3)",
        "FQBN": "stm32duino:STM32F1:maple",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103V series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103V",
        "VID": "0x1EAF",
        "PID": "0x0004"
      }
    ]
  }
]
```

**To Reproduce**
Steps to reproduce the behavior:
1. Add additional boards URL "https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json"
2. Install STM32 core
3. attach board (I am using Generic STM32F103RET6)
4. Look at ports or run CLI list

**Expected behavior**
If multiple uC are using the same PID/VID then should it be somthing Generic and Unique from the CLI like "STM32F103 Series" 

**Desktop (please complete the following information):**
 Windows 10

ubidefeo, 2021-03-16:
@toogooda 
the issue lies in the STM32 core having multiple aliases for the same `VID:PID` pair.
The CLI is doing a serial discovery operation which matches said pair against all the ones defined in the various cores.

Arduino boards have custom `VID:PID` pairs (each product has one ore more) so the match is pretty much always exact.
That cannot be done when there's no native, configurable (and configured) USB descriptor with proper `VID:PID` markers.
I think these boards all use the same USB Serial chip or do it natively so it's stored as a fix setting in the bootloader.
Not sure how it gets implemented on each board, but I know for sure that the Maple does not have a USB<>Serial chip on board

per1234, 2021-03-16:
>Com port selection for IDE is not working

Please provide a detailed description of what you mean by "not working" @toogooda.

I just tried compiling and uploading for a board under these conditions it functioned perfectly well for me.

Is it only an aesthetic issue you're reporting?

>If multiple uC are using the same PID/VID then should it be somthing Generic and Unique from the CLI like "STM32F103 Series"

I don't think that would be correct. The redundant VID/PID might only be used for a subset of the boards of the platform.

toogooda, 2021-03-18:
Hi, After more testing it turns out that a very old version of Dan Browns STM32 core which was never installed on IDE 2.0 was what was mapping all those boards to the same VID/PID and making the IDE very confusing. Once uninstalled it was all working fine.

closed 2021-05-11 by silvanocerza