arduino/arduino-cli #1227: 404 Not Found errors for task build (unrecognized import path) - downloader/v2

T-vK, 2021-03-18:
## Bug Report

I wanted to install `arduino-cli` from source. For that I followed the instructions here: https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#prerequisites to the point where I'm supposed to run `task build`.

### Current behavior

`task build` fails.

``` Bash
task build --verbose
go: go.bug.st/downloader/v2@v2.1.1: unrecognized import path "go.bug.st/downloader/v2": reading https://go.bug.st/downloader/v2?go-get=1: 404 Not Found
task: dynamic variable: 'echo `go list ./... | grep -v legacy | tr '\n' ' '`' result: ''
task: dynamic variable: 'echo "$(git log -n 1 --format=%h)"' result: 'ab92281a'
task: dynamic variable: 'echo "$(date -u +"%Y-%m-%dT%H:%M:%SZ")"' result: '2021-03-18T12:30:03Z'
task: dynamic variable: 'echo "20210318"' result: '20210318'
task: dynamic variable: 'echo "`git tag --points-at=HEAD 2> /dev/null | head -n1`"' result: ''
go: go.bug.st/downloader/v2@v2.1.1: unrecognized import path "go.bug.st/downloader/v2": reading https://go.bug.st/downloader/v2?go-get=1: 404 Not Found
task: Command "go list -f {{".Target}}" golang.org/x/lint/golint" in taskvars file failed: exit status 1
```

### Expected behavior

`task build` works without errors.

### Environment

- CLI version (output of `arduino-cli version`): master @ ab92281a7904019d9de8b9dd74964e0dbf106a5c
- OS and platform: Fedora 32 x64

### Additional context

I also tried installing `protoc-gen-go`, even though my understanding is I wouldn't need it
```
go get -v -u github.com/golang/protobuf/protoc-gen-go
go: go.bug.st/downloader/v2@v2.1.1: unrecognized import path "go.bug.st/downloader/v2": reading https://go.bug.st/downloader/v2?go-get=1: 404 Not Found
```

mlausch1963, 2021-04-19:
I was hit by the same problem, and I opened a bug at https://github.com/bugst/go-downloader/issues  but no reaction so far.
The problem is in the github.com/bugst/go-downloader/ module file they have `module go.bug.st/downloader/v2`
so even if I do a local change to the "real" module path, without redirections, it works only as long as the go.mod file is not fetchd.

cmaglie, 2021-06-08:
Sorry for the long delay, for some reason this issue went off my radar, the problem should be fixed now.

Amazingly, in my local system, everything is working fine even before the fix...