arduino/arduino-cli #1229: config file to accept command options

RalphCorderoy, 2021-03-21:
It's tedious writing `--fqbn ...`. I have a arduino-cli.yaml in the current directory but it only accepts a few options. A _cmd_ section could have have subsections for each subcommand and within those each would be a subcommand's option, e.g. _cmd.compile.fqbn_. This would allow per-sketch settings to be stored under Git unlike environment variables. And if a file is sourced to set the environment then forgetting to do it causes problems whereas the local arduino-cli.yaml is always present.

ubidefeo, 2021-03-22:
@RalphCorderoy 
we are addressing this need in other ways.
The CLI config file is not the correct place for this to happen.
For now you have to go through this it's maybe faster to create a "build file" you can pass parameters to and wrap the CLI.

A great example is how Sayanee Basu uses Arduino CLI with a Makefile
https://www.youtube.com/watch?v=cVod8k713_8

In the description you also find links to the demos she runs.
In the future we'll introduce easier ways of getting around this

jockm, 2021-05-12:
Sure, but one day you need to do something unusual and different and have to use the command line and then I have to manually enter a 143 character FQBN just do do something one off.  The command line is all well and good, but I would like a settings file for build options... like other similar platforms do

RalphCorderoy, 2021-05-14:
Thanks, @ubidefeo, but I already have a Makefile with targets to build the hex, tags, prog, clean, etc., all of which can use the Makefile's $(BOARD_NAME) I've set when running arduino-cli. The problem comes, as @jockm says, when wanting to do ad hoc arduino-cli runs at the shell rather than the canned Makefile commands, perhaps because some different options want to be given.

silvanocerza, 2021-05-14:
There are plans to store some information in the `sketch.json` file to run the `compile` command without the need to specify some flags, nothing set in stone for now but it will come sooner or later depending on what we'll need to prioritize. Anyway we need to work on some other things before working on that.

It would greatly help if you outlined your workflow, maybe there is already a way to do what you want to do but I need more information to be sure.

On a side note the `arduino-cli.yaml` file is used to store global configs for the Arduino CLI only, it's not meant to store individual Sketch configurations.

jockm, 2021-11-26:
@silvanocerza If the arduino-cli.yaml can live in the sketch directory, then conceptually it isn't just global configuration.  There is no practical reason it can't be used for sketch level settings.  Indeed I have always assumed there was global one and the local one would override the specific settings present.  

Putting sketch level settings there makes sense, without the need to add another file to the directory

silvanocerza, 2021-11-26:
>  Indeed I have always assumed there was global one and the local one would override the specific settings present.

You assumed correctly, if the `arduino-cli` finds a `arduino-cli.yaml` file in the current working directory it overrides the global one. I don't like this behaviour and would very much prefer to change it, `arduino-cli.yaml` must be meant to store settings about the `arduino-cli` and only that.
It now works like this for legacy reasons, it will probably change in the future.

> Putting sketch level settings there makes sense, without the need to add another file to the directory

We have a plan to store Sketch configurations, part of that plan is outlined in the [build profiles RFC](https://github.com/arduino/tooling-rfcs/blob/main/RFCs/0003-build-profiles.md). 
`arduino-cli.yaml` as of now is not meant to store any Sketch configurations and probably never will.

jockm, 2021-11-26:
@silvanocerza 

>  as of now is not meant to store any Sketch configurations and probably never will.

Yes that was clear from what was previously said in this issue.  I was arguing that you don't need another file.  I was giving feedback

per1234, 2021-11-26:
>If the arduino-cli.yaml can live in the sketch directory

It can, but note there is no special treatment of the configuration file in that location. The configuration file is handled the same when it is in the sketch folder as when it is in any other folder. 

Other parts of Arduino CLI do give special treatment to the sketch folder in that it uses the current working directory as the default sketch path argument. So it is convenient to run Arduino CLI commands from the sketch folder, and when you do that a configuration file in the sketch folder will be used. But Arduino CLI commands also might be ran from any arbitrary location, specifying the sketch path as an argument to the command.

It is possible that this loading of the configuration file from the current working directory might be problematic: https://github.com/arduino/arduino-cli/issues/758

umbynos, 2022-11-30:
Hi @RalphCorderoy, we added the [sketch project file](https://arduino.github.io/arduino-cli/dev/sketch-specification/#sketch-project-file) which should solve your problem.
The command `arduino-cli board attach -b <fqbn> <sketchpath>` will produce a `sketch.yaml` file that will contain the fqbn to be used as default.

closed 2022-11-30 by umbynos