arduino/arduino-cli #1238: needs cli ??? what ?

phil123456, 2021-03-27:
## Bug Report

I cloned the repo on a raspberry pi 4 and did install.sh
and it complains it does not find....arduino-cli

### Current behavior

pi@raspberrypi:~/arduino-cli $ sudo ./install.sh 
Installing in /home/pi/arduino-cli/bin
ARCH=ARMv7
OS=Linux
Using curl as download tool
TAG=0.17.0
CLI_DIST=arduino-cli_0.17.0_Linux_ARMv7.tar.gz
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.17.0_Linux_ARMv7.tar.gz
arduino-cli not found. You might want to add /home/pi/arduino-cli/bin to your $PATH
arduino-cli alpha Version: 0.17.0 Commit: ab587583 Date: 2021-03-17T15:10:16Z installed successfully in /home/pi/arduino-cli/bin

### Expected behavior

to install

### Environment

- CLI version (output of `arduino-cli version`): none
- OS and platform: raspbian
5.4.51-v7l+

### Additional context

I want to run arduino command from the command line since the arduino executable cannot to this anymore apparently

phil123456, 2021-03-27:
I added the path so it finds angular-cli
but why do I even bother with the repository at all ???

per1234, 2021-03-27:
>it complains it does not find....arduino-cli

I think you have since realized that this message was telling you that the installation location was not in your `PATH`. The reason you might want it in your path is so you can simply run `arduino-cli` from any location. However, it's perfectly fine to install Arduino CLI outside the path. This only means that you will need to specify the full path to the executable to use it.

It seems like maybe the wording of this message could be made more clear.

>but why do I even bother with the repository at all ???

I don't know. The installation instructions seem to be clear about how to install Arduino CLI:
https://arduino.github.io/arduino-cli/latest/installation/
If you have any suggestions for improvements for those instructions, you're welcome to submit a pull request.

atannen, 2024-01-16:
this may be a stale bug report, but I just ran into this quirk.  consider:

I have arduino-cli on a few Linux PCs, and I wrote my own get-arduino-cli wrapper script
that calls the install.sh and then symlinks arduino-cli to the current arduino-cli_ver.
so, I call my get-arduino-cli remotely with pdsh, and it was giving me:

   arduino-cli not found. You might want to add "<my $HOME/bin>" to your $PATH

OK, that was true, because the pdsh remote isn't reading my .profile
(my .profile adds $HOME/bin to my $PATH.)
but I didn't even know where the error message was coming from,
or why arduino-cli was being called without an absolute path.
(because I wasn't calling it, the install.sh was.)

a better warning message would be:

    install.sh: arduino-cli not found. You might want to add "<my $HOME/bin>" to your $PATH

at least I'd know it's not my script that is expecting to find arduino-cli in the $PATH.
So I hacked my wrapper script to make sure that $HOME/bin is in my $PATH,
though it's only necessary to fix this wrinkle, so the next person who runs my script
doesn't get confused like I did.

closed 2024-03-20 by MatteoPologruto