arduino/arduino-cli #1253: ) in comment breaks function prototype generation

FRautenberg, 2021-04-03:
If a closing bracket is commented out within parameters, no prototype is generated for the function.

Compiling this sketch:
```c++
void setup() {
  foo(1,2);
}
void loop() {}
void foo(int a,  // )
         int b) {}
```
Fails:
```
C:\Users\per\AppData\Local\Temp\arduino_modified_sketch_281391\sketch_apr03a.ino: In function 'void setup()':
sketch_apr03a:2:3: error: 'foo' was not declared in this scope
   foo(1,2);
   ^~~
```
If the `)` is removed from the comment the prototype generation works correctly:
```c++
void setup() {
  foo(1,2);
}
void loop() {}
void foo(int a,  //
         int b) {}
```
Verified with:
- Arduino IDE 1.8.14 Hourly Build 2021/03/09 09:33
- Arduino CLI nightly-20210401 Commit: a4ee670a

Windows 10

ubidefeo, 2021-04-07:
please @cmaglie 
see if this is resolved or not

per1234, 2021-04-07:
@ubidefeo it's not resolved. I verified it with the latest nightly Arduino CLI before transferring it here from the `arduino/Arduino` repository.

ubidefeo, 2021-04-07:
@per1234 I have assigned it to @cmaglie based on your previous comment 
>I'll leave it to the developers to decide whether to consider this bug fixed already.

umbynos, 2022-10-12:
This seems to be a problem of the preprocessor (ctags) fixing it upstream is basically impossible, an alternative would be to switch to arduino-prepocessor but needs some love (still uses an old version of llvm). I don't see it coming shortly.