arduino/arduino-cli #1260: [feature-request][grpc] Make the `daemon.port` optional

kittaakos, 2021-04-12:
It would be great if the gRPC server for the CLI daemon could start on any available port if the `daemon.port` is not defined in the CLI configuration. So when the following is not specified:
```yaml
daemon:
  port: "50051"
```
The gRPC server would start on a random port, the new IDE could read the port from the stdout, and could connect to it.

https://github.com/arduino/arduino-cli/blob/a4ee670a9233a107bb89d4cfe0c48ab40aba9076/cli/daemon/daemon.go#L124

Downstream request: https://github.com/arduino/arduino-ide/issues/201

Thank you!

per1234, 2021-12-09:
@silvanocerza @fstasi do you think there is any need for the requested capability now that it is possible (https://github.com/arduino/arduino-cli/pull/1585) to take the port 0 approach, as done in https://github.com/arduino/arduino-ide/pull/673?

silvanocerza, 2021-12-09:
Yes, this can be closed. 
If the user wants a random port she can either set the `daemon.port` config to `0` or use the `--port` flag.

fstasi, 2021-12-09:
@per1234 I confirm we can close this one

closed 2021-12-09 by per1234