arduino/arduino-cli #1262: [grpc] `lib list --fqbn the:board:fqn --all` throws incorrect error when `directories.data` folder does not exist at daemon startup

kittaakos, 2021-04-12:
## Bug Report

As a gRPC consumer of the CLI, I receive an incorrect error message from `lib list` when the `directories.data` location does not exist before the `InitRequest`.

### Current behavior

I cannot attach code, but this is my workflow:
 - Clean `directories.data` location (on macOS is it `rm -rf ~/Library/Arduino15/`),
 - Create a client with the `InitRequest`.
 - Note: at this point, the `Arduino15` folder exists on my filesystem:
```sh
Arduino15 % ls -al
total 8
drwxr-xr-x   6 akos.kitta  staff   192 Apr 12 17:02 .
drwx------@ 92 akos.kitta  staff  2944 Apr 12 17:02 ..
-rw-r--r--   1 akos.kitta  staff   104 Apr 12 17:02 inventory.yaml
drwxr-xr-x   3 akos.kitta  staff    96 Apr 12 17:02 packages
drwxr-xr-x   3 akos.kitta  staff    96 Apr 12 17:02 staging
drwxr-xr-x   2 akos.kitta  staff    64 Apr 12 17:02 tmp
```
 - Run a `LibraryListRequest`
```json
{
  "instance": { "id": 1 },
  "all": true,
  "updatable": false,
  "name": "",
  "fqbn": "arduino:samd:arduino_zero_edbg"
}
```
 - It fails, but with an incorrect error message:
```
2 UNKNOWN: loading board data: unknown package arduino
```

### Expected behavior

 - The `LibraryListRequest` fails, but with the following error message:
```
2 UNKNOWN: loading board data: platform arduino:samd is not installed
```

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli alpha Version: nightly-20210408 Commit: a4ee670a Date: 2021-04-08T01:26:00Z`
- OS and platform:

### Additional context

 - I had my Arduino Zero board attached.
 - I used the default CLI config with two additional 3rd party URLs.
 - I tried to run a `RescanRequest` right after the `InitRequest` and before the `LibraryListRequest`. The `RescanResponse` was completely broken on my side: it had one library and multiple platform errors.

kittaakos, 2021-04-12:
The same error happens when I do `board details` instead of `lib list`.

kittaakos, 2022-11-30:
I do not know if this is still valid, but I checked, and the IDE2 code still has a guard against it:
 - https://github.com/arduino/arduino-ide/blob/d6a4b0f910e37c3ecd569ca457f91b577734e4df/arduino-ide-extension/src/node/boards-service-impl.ts#L551-L554
 - https://github.com/arduino/arduino-ide/blob/d6a4b0f910e37c3ecd569ca457f91b577734e4df/arduino-ide-extension/src/node/library-service-impl.ts#L186-L190

kittaakos, 2023-01-23:
This issue works as expected when using the CLI from a terminal:

```
./arduino-cli version
arduino-cli  Version: git-snapshot Commit: a58d5adb Date: 2023-01-23T09:37:23Z
```

```
rm -rf ~/Library/Arduino15
rm -rf ~/Documents/Arduino/libraries
./arduino-cli lib list -b arduino:samd:arduino_zero_edbg --format json
{
  "error": "Error listing Libraries: Unknown FQBN: platform arduino:samd is not installed"
}
```

```
rm -rf ~/Library/Arduino15
rm -rf ~/Documents/Arduino/libraries
./arduino-cli board details -b arduino:samd:arduino_zero_edbg --format json
{
  "error": "Error getting board details: Unknown FQBN: platform arduino:samd is not installed"
}
```

----
When using the gRPC APIs, the libraries and platform index must be updated before the `InitRequest`. This is precisely what IDE2 does here. Otherwise, the core client is not functional. Steps with grpcurl:

```
grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Create
```

```json
{
  "instance": {
    "id": 1
  }
}
```

```
grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 1}}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.UpdateIndex
```

```json
{
  "downloadProgress": {
    "start": {
      "url": "https://downloads.arduino.cc/packages/package_index.tar.bz2",
      "label": "Downloading index: package_index.tar.bz2"
    }
  }
}
{
  "downloadProgress": {
    "update": {
      "downloaded": "45861",
      "totalSize": "45861"
    }
[... 368 lines cut ...]
 - `Arduino SAMD Boards (32-bits ARM Cortex-M0+) [v 1.8.13]` is installed,
 - Select the Arduino Zero (Programming Port),
 - Quit IDE2,
 - Delete `directories.data` and `directoties.user/libraries`.

The caught exceptions on the IDE2 side do not contain this `unknown` error message anymore. Errors were:

```json
{"code":9,"details":"Platform 'arduino:samd' not found: platform arduino:samd is not installed","metadata":{"content-type":["application/grpc"]}}
```

```json
{"code":5,"details":"Unknown FQBN: platform arduino:samd is not installed","metadata":{"content-type":["application/grpc"]}}
```

The (gRPC) commands where
 - lib list -b
 - board details -b
 - (and getting the user fields)

-----
 - I do not know why the responses are not in JSON.
 - I do not know why there is an empty object (`{}`) during the update progress.

Feel free to close it.

Bikappa, 2023-01-23:
@kittaakos regarding the error formatting, that comes from `gprcurl`. There is a dedicated `-format-error` option
 Try:
 ```
 grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 1}, "fqbn": "arduino:samd:arduino_zero_edbg"}' \
  -format-error \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.LibraryList
```
I'll create an issue for each of the other two points

closed 2023-01-23 by Bikappa